bigtreetech/CB1 #2: Octoprint + CB1?

Aleks-MO, 2022-07-22:
Would you like to see the installation of **Octoprint** on **CB1** ?

escvn54, 2022-07-23:
I would definitely like to see this.  Even better, a barebones CB1 Ubuntu image with nothing installed, so the end user can decide what to install.

astrof, 2022-07-26:
I was under the impression the reason for these was for Klipper and octoprint

escvn54, 2022-07-26:
> I was under the impression the reason for these was for Klipper and octoprint

You're absolutely right, but I would rather have the clean slate to start with, so I can install only what I want installed.

astrof, 2022-07-26:
So obviously the correct answer is...have both! Have a clean image version and one that has the 3D Printer tools (and/or whatever else normal BTT customers use) installed (or instructions on how to install)

bigtreetech, 2022-07-30:
Hello. Every one. The minimal Debian is released in [here](https://github.com/bigtreetech/CB1/releases).  And we can discuss the optimization and improvement [here](https://github.com/bigtreetech/CB1/discussions/9)

Will-wastelander, 2022-10-11:
I second a vanilla image. What if I don't plan to use the CB1 for 3D
Printing tasks, and want to use it for something else m

closed 2024-03-07 by bigtreetech