bigtreetech/CB1 #4: Ubuntu 21.10 No Longer Supported

escvn54, 2022-07-25:
As of July 14th, Ubuntu 21.10 is no longer supported.  Recommend building your images off of an LTS build.

astrof, 2022-07-26:
One idea is could also go off of Armbian (as Raspian/Octoprint image does) or do a fork of CustomPiOS

escvn54, 2022-07-26:
> One idea is could also go off of Armbian (as Raspian/Octoprint image does) or do a fork of CustomPiOS

I completely agree, if I were savvy enough, or had enough time, I would commit to doing that. I just don't see that happening.  As it is, I'm stuck with an outdated (ever so slightly) OS.

astrof, 2022-07-26:
So it shouldn't be that difficult; CustomPiOS is literally built for this:
https://github.com/guysoft/CustomPiOS
It even has a link on how to build off of Ubuntu so we could start with that.  And then octoprint uses that to install off of as well so double win.

SHoover80, 2022-07-26:
So what does one need to do to get a working OS on a CB1 as I just just got it, and trying to use what appears to be the latest available OS and it's out of date..........

emesix, 2022-07-28:
Is the hardware not the same as a Orange Pi zero 2 ? So should a linux image compatible with the Allwinner H616 /  Orange Pi zero 2 work out of the box?

SHoover80, 2022-07-28:
I tried it. It wouldn't boot.

MarkusThur, 2022-07-28:
Pretty same here.. I just stupidytried to boot a octopiorbetter the octoklipperpi boot image.. definitely did not boot.

I'll try the OS provided here to get it boot, but am not sure how to copy it on the SD

midicdj1000, 2022-07-29:
HI, so this is a big problem for me at the moment. I have tried to install Klipperscreen and it will not install. I tried to update Moonraker and it will not update.

escvn54, 2022-07-29:
@bigtreetech 

Can you please update the image to 22.04, or another LTS release.

Thank You!

SHoover80, 2022-07-29:
Good luck trying to get a response from @Bigtreetech, @escvn54. They keep releasing updated versions of the OS that are all running on an outdated OS.

escvn54, 2022-07-30:
@SHoover80 , it looks like there's a barebones Debian system to try out. I'm going to give it a shot and see if it at least has the ability to update/upgrade.

bigtreetech, 2022-07-30:
Hello. Every one. The minimal Debian is released in [here](https://github.com/bigtreetech/CB1/releases).  And we can discuss the optimization and improvement [here](https://github.com/bigtreetech/CB1/discussions/9)

SHoover80, 2022-07-30:
Yup, I see that as of like an hour ago @escvn54. I'm happy now. I'm able to play around with it and dive into and learn Klipper. Finally, a functioning OS. Setting it up now. I was able to install with no issues, update and access the UI. The only trouble I'm having is trying to get my USB camera working. Just ignorance on my part learning Klipper. I'll figure it out though now that I can eliminate out of date OS being the cause.

Will-wastelander, 2022-10-11:
I may give this a try. I'll keep you guys posted.

closed 2024-03-07 by bigtreetech