bigtreetech/CB1 #5: Unable to get a pi camera to work. Is this the only OS that'll work on this board?

SHoover80, 2022-07-26:
Been having issues getting the CB1 to work with the Bigtreetech Pi4B adapter and I cant get the camera to work. I get nothing out of the HDMI port.  
I've tried to run other versions of Pi and nothing happens. The Pi4B adapter shows no LED's for power or activity and had no indication it was working other than once I tried this OS I was able to see the Ethernet port lights light up. but I get absolutely nothing when trying other versions of Klipper or even octopi. 
I've tried contacting BTT support but that's a joke. I dont think even though they sell a lot in the states that they have anyone that comprehends English in their support department as it's like speaking with a bot when emailing their support as their responses are oblivious to the issues I requested help with so I figured I'd try through here.

frutabruta, 2022-07-26:
1. The camera (CSI) and display ports (DSI) are not supported on the CB1. It is even stated in the listing of the item. It works only on the CM4 plugged into the adapter
2. You need to have operating system specifically compiled for processor Allwinner H616. So operating system images compiled for original Raspberry Pi's will not work on CB1.

SHoover80, 2022-07-26:
I appreciate the response but I cant really do anything with it now due to not being able to upgrade

frutabruta, 2022-07-26:
The Hurakan_0711.img.xz version works for me, it's based on recent Debian. It even shows console on the HDMI (only one of the two HDMI works). If you want to use klipperscreen, it needs a patch installed from BTT version of kiauh. ( whole process described in Manta M8P manual on github). Unfortunately, I could't get git to work on CB1 (git pull ends on timeout), so I had to download the kiauh on different computer and transfer it to CB1.

SHoover80, 2022-07-26:
I'll give the Hurakan_0711.img a try. I was using Hurakan_20220715, then troubleshooting trying to figure out why I couldn't get the camera to work, I had the Hurakan_20220715 working and updating. Seen the Hurakan_20220722_4.9 was out and gave it a try. But it would tell me It needed to update and when in SSH trying to update I just get: 

Ign:1 http://ports.ubuntu.com/ubuntu-ports impish InRelease
Ign:2 http://ports.ubuntu.com/ubuntu-ports impish-updates InRelease
Ign:3 http://ports.ubuntu.com/ubuntu-ports impish-backports InRelease
Ign:4 http://ports.ubuntu.com/ubuntu-ports impish-security InRelease
Err:5 http://ports.ubuntu.com/ubuntu-ports impish Release
  404  Not Found [IP: 185.125.190.36 80]
Err:6 http://ports.ubuntu.com/ubuntu-ports impish-updates Release
  404  Not Found [IP: 185.125.190.36 80]
Err:7 http://ports.ubuntu.com/ubuntu-ports impish-backports Release
  404  Not Found [IP: 185.125.190.36 80]
Err:8 http://ports.ubuntu.com/ubuntu-ports impish-security Release
  404  Not Found [IP: 185.125.190.36 80]
Reading package lists... Done
E: The repository 'http://ports.ubuntu.com/ubuntu-ports impish Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports impish-updates Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports impish-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ports.ubuntu.com/ubuntu-ports impish-security Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I tried going back to the Hurakan_20220715 version and get the same thing then when trying to go back into Fluidd it  loose access only getting the options to refresh or force refresh.  While SSH I can't install the Kiauh scripts without getting errors unable to download. when I was able to yesterday.

SHoover80, 2022-07-27:
I followed your advice, and it's updating now. 
Thank you @frutabruta

SHoover80, 2022-07-27:
Well I spoke too soon. I tried to install Kiauh and got an error showing a local IP that's not in the range of my local network  that started with 192 and some port 7??? So I figured I'd powercycle the CB1+Pi4B and it's not connecting to my network, No LED's on the LAN port that would previously light up when it was running since someone was too lazy to implement the Acc/power LED. so one cant determine if it's running or even booting. A couple more power cycles and no sign of life. Re flashing the MicroSD and still no signs of life.

escvn54, 2022-07-27:
This looks to be the same chipset and hardware as the OrangePi Zero 2.  I would be interested to see if Armbian would boot up on it with some work, as there's an Armbian build for the Zero 2, which I also own.  I may take a look at that.  

As it stands, my USB camera works fine, and as mentioned before by frutabruta, the CB1 doesn't support the DS or CS ports, which is really a shame, considering BTT's own Raspberry Pad and the CM4 adapters are really set up well for this product, and the CB1 has the horsepower to handle it.  Would be nice to see a future firmware update that might fix that, if possible.

frutabruta, 2022-07-27:
[SHoover80](https://github.com/SHoover80) Local IP: that is the problem I had with git clone:
` Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/bigtreetech/kiauh.git/': Failed to connect to 192.168.0.126 port 7890: Connection timed out `

But my local network is 192.168.1.x , so it does not make any sense. I got the same error both on Ubuntu based image and on the Raspbian based Image.
Not connecting to network/not booting: That's weird. Mine reboots normally and even connects to Wi-Fi, but I am not using ethernet right now. The problem might be a bad USB charger or bad SD card.

apt errors: the Ubuntu verion is based on Impish Indri version, which is no longer supported by Ubuntu and therefore the repositories it is looking for are no longer present on the Ubunu servers.

[escvn54](https://github.com/escvn54) :  the Hurakan_0711.img.xz  version boots with OrangePi logo, so I guess that's where they started from. I tried to download the unmodified version for OrangePi Zero 2, but it does not boot/show anything on HDMI.

CB1 not supporting DSI or CSI is a hardware property. BTT mentioned in comments on Facebook that these ports will be functional with the new module called CB2. An no, RaspberryPad is not compatible with CB1 either. Again, confirmed by BTT.

SHoover80, 2022-07-27:
Yea, I'm not sure what's going on with it. It appears for whatever reason the LEDs on the ethernet port just quite working. I won't even light up when connecting to an ethernet port. I did find a cable to connect the HDMI and the one I was using was bad. I did see the orange pie logo on the screen. and I was finally able to log into through SSH. and Fluid. and now with the monitor connected I'm able to see rc.local [19745]: ping: SO_BINDTODEVICE eth0: no Such Device repeatedly. so, I take it it's not detecting the ethernet port. I was able to do some updates with Wi-Fi but when using Kiauh.sh It argues with me: 
[image]

[image]

Every time I get past one point, I end up with more issues dealing with this. I was hoping to have it ready for my MantaM8P board that'll be here tomorrow but looks like I'm just going to put this CB1 in it's place (trash) and get a CM4. Cost more but works.

frutabruta, 2022-07-27:
[SHoover80](https://github.com/SHoover80)
I found the solution to the git problem :)
Run these commands:

```
git config --global --unset https.proxy
git config --global --unset http.proxy
```

SHoover80, 2022-07-28:
[image]
Who's to blame? Hardware manufacturers? some of the issues could be hardware related. The fact that the LED's aren't functioning indicates basic things like power, activity. How the Ethernet port suddenly quit working. SO, they have some faults. Software developers, most definitely share fault in this poor excuse of a product BTT is selling. Trying to piece together software meant for other hardware and going out of their way to hide the fact that it's Orange Pi hardware and obviously the software isn't working out of the box. All the hoops one must go through, modifying the out of box configs for the OS intended for this device just to get it to update which is required for full functionality.  Yet once updated you can't access the UI rendering this useless and a waste of my time/money. Especially going towards a company that refuses to support its products and relies on its customers to do the support for free. I'm fucking done with this shit. I got too much to do and wasted too much god damn time trying to get this piece of shit working.

frutabruta, 2022-07-28:
LED's: in the Ubuntu version, the LED blinks to indicate running state.
Ethernet quit working: I think it has something to do with the Armbian vs. Ubuntu version. I can't even see eth0 network card in ifconfig in Armbian.
I feel sorry for you that you expected a product out of the box, the support for custom Linux PCs is almost always terrible. I was already having similar experience with OrangePi PC and I hope the CB1 will be saved by the same way: Official vanilla Armbian release without any modifications.
Klipper workaround: use kiauh (the offcial one, not BTT branch) to remove component one by one (mainsail, moonraker, klipper) and reinstall them. Then errors should go away.

SHoover80, 2022-07-29:
@Frutabruta, I do expect something I buy to work out of the box. at-least a OS out of the box that boots and updates.

bigtreetech, 2022-07-30:
Hello. Every one. The minimal Debian is released in [here](https://github.com/bigtreetech/CB1/releases).  And we can discuss the optimization and improvement [here](https://github.com/bigtreetech/CB1/discussions/9)