bigtreetech/CB1 #6: git clone does not work

frutabruta, 2022-07-27:
Every git clone end up with 
`
'Failed to connect to 192.168.0.126 port 7890: Connection timed out'
`

This IP is specified in .gitconfig file:
```
biqu@Hurakan:~$ ack "192.168.0.126"
.gitconfig
2:      proxy = http://192.168.0.126:7890/
4:      proxy = https://192.168.0.126:7890/
```
Which  can be resolved with two commands:

```
git config --global --unset https.proxy
git config --global --unset http.proxy
```

emesix, 2022-07-27:
Try this: git config --global --unset http.proxy

emesix, 2022-07-28:
Dear BigTreeTech,

Is there anyway to get a empty minimal install of Debian for the CB1? The last ISO image is full of proxy settings, personalized shell configurations and preinstalled non release software. Compiling the klipper software for the M8P fails on a normal SSH login because of missing en_US locale!

bigtreetech, 2022-07-29:
> Dear BigTreeTech,
>
> Is there anyway to get a empty minimal install of Debian for the CB1? The last ISO image is full of proxy settings, personalized shell configurations and preinstalled non release [...]

The Debian11 file system is being ported. The minimal OS of Debian version will be released before tomorrow. Thank you for your support

bigtreetech, 2022-07-30:
> Dear BigTreeTech,
>
> Is there anyway to get a empty minimal install of Debian for the CB1? The last ISO image is full of proxy settings, personalized shell configurations and preinstalled non release [...]

Hello. The minimal Debian is released in [here](https://github.com/bigtreetech/CB1/releases).  And we can discuss the optimization and improvement [here](https://github.com/bigtreetech/CB1/discussions/9)

frutabruta, 2022-09-28:
This issue has been solved in new images.

closed 2022-09-28 by frutabruta