bigtreetech/CB1 #10: Debian image CB1_Debian11_bullseye_minimal_kernel4.9_20220730.img

emesix, 2022-07-30:
APT source list are still some local setting in the CB1 Debian image.

sudo nano /etc/apt/sources.list
deb https://mirrors.tuna.tsinghua.edu.cn/debian bullseye main

But rest of the image is pretty clean!!! Thank you Biqu / BigTreeTech

emesix, 2022-07-30:
Biqu - user home directoy has a folder:
root root 4096 Jul 30 12:21 gcode_files 
Remove folder before installing or change owner.
sudo chown -R biqu gcode_files/
sudo chgrp -R biqu gcode_files/

escvn54, 2022-07-30:
That's the same source used by the OrangePi debian build, so no surprise, as it seems like BTT has built their release on top of the OrangePi Zero 2 release of Debian or Ubuntu respectively.

escvn54, 2022-07-30:
If you would prefer non CN sources, as I do:

deb http://deb.debian.org/debian bullseye main
deb-src http://deb.debian.org/debian bullseye main

deb http://deb.debian.org/debian-security/ bullseye-security main
deb-src http://deb.debian.org/debian-security/ bullseye-security main

deb http://deb.debian.org/debian bullseye-updates main
deb-src http://deb.debian.org/debian bullseye-updates main

escvn54, 2022-07-30:
@emesix , did you figure out the root/su password?

emesix, 2022-07-30:
Yes it has the same hardware as a Orange Pi Zero 2. But i just want to make clear that BTT should remove all there references in OS images. 

Biggest problem i have now is that ALLWINNER H616 is supported in klipper to be used for the motion sensor adxl345 ? And i cant get /tmp/klipper_host_mcu working...

midicdj1000, 2022-07-31:
So far this has worked for me, well at least installing Klipper, Mainsail, Moonraker and Klipper Screen worked without issue, The firmware  compiled and Klipper can see the MCU so this is progress over the first Image.

djscrubb, 2022-07-31:
> Yes it has the same hardware as a Orange Pi Zero 2. But i just want to make clear that BTT should remove all there references in OS images.
>
> Biggest problem i have now is that ALLWINNER H616 is [...]

Have fun enabling spi bricked my image

Aleks-MO, 2022-07-31:
There is no Wi-Fi connection..... I can't configure it. (((((

frutabruta, 2022-08-01:
> There is no Wi-Fi connection..... I can't configure it. (((((

Did you enter the Wi-Fi SSID and password to the file on microSD card? It is on one of the partitions mounted on Windows after writing the image. I think the file is called system.cfg.

Aleks-MO, 2022-08-01:
frutabruta. No. I don't see this disk as BOOT. So it was not possible to edit the system.cfg file ....(((( How to force it through an SSH connection?
[biqu@CB1 ~]$ iw wlan0 link
Not connected.
[biqu@CB1 ~]$  sudo ip link set wlan0 up
[biqu@CB1 ~]$ iw wlan0 link
Not connected.
[biqu@CB1 ~]$ sudo iwlist wlan0 scan | grep ESSID
                    ESSID:"Taxi-GSM-VIP"
                    ESSID:"TP-LINK_C5D2CC"
                    ESSID:"DIRECT-JR-BRAVIA"
                    ESSID:"DIRECT-UA-BRAVIA"
[biqu@CB1 ~]$ nmcli d wifi connect Taxi-GSM-VIP password ***************
Error: Failed to add/activate new connection: Not authorized to control networking.
????????????????????

frutabruta, 2022-08-01:
@Aleks-MO Not sure about this release, but I think it's mounted as /mnt/u-disk or /media/u-disk.

Aleks-MO, 2022-08-01:
@frutabruta  It doesn't matter anymore, this system is not visible on my computer and it is impossible to edit. I don't know how to start Wi-Fi now...

midicdj1000, 2022-08-01:
Hi @Aleks-MO what size SD card are you using ? I had problems with a brand new 64gb sd card, I had the same problem that it wouldn't show any valid partitions on my pc. I switched to a smaller 8gb and it worked first time I at the end the drive with the wifi config opened and I could edit it.

Aleks-MO, 2022-08-01:
Hi! @midicdj1000  SD 32/16/8gb sd card..

Aleks-MO, 2022-08-01:
@midicdj1000 In the current state, is it possible to start WiFi somehow?

midicdj1000, 2022-08-01:
are you using the raspberry pi image tool ? I stopped it trying to write the wifi and ssh stored settings.

midicdj1000, 2022-08-01:
perhaps clean the 8gb sd first and try again.

Aleks-MO, 2022-08-01:
Yes, raspberry pi image tool

djscrubb, 2022-08-01:
I used belina etcher and had to problems. This image for what ever reason doesn't use the system.txt file

midicdj1000, 2022-08-01:
How would I setup HDMI screen sizes ?

Aleks-MO, 2022-08-01:
You can use Win32DiskImager, the result is the same...

billru05, 2022-08-29:
> If you would prefer non CN sources, as I do, you can edit your sources.list with the following sources:
> 
> deb http://deb.debian.org/debian bullseye main deb-src http://deb.debian.org/debian bullseye main
> 
> deb http://deb.debian.org/debian-security/ bullseye-security main deb-src http://deb.debian.org/debian-security/ bullseye-security main
> 
> deb http://deb.debian.org/debian bullseye-updates main deb-src http://deb.debian.org/debian bullseye-updates main

TxBillbr, 2022-09-02:
BTT needs to comply with the license and post the sources. There are many people who could help quickly get the image into a better shape.  They also need to contact Debian to get into their ARM program.  (https://www.debian.org/ports/arm/) and (https://wiki.debian.org/InstallingDebianOn/Allwinner)