bigtreetech/CB1 #7: GPL license violation - Missing source release

pendor, 2022-07-30:
GPL requires sources necessary to fully rebuild a binary release be published as part of the release.  The "source" tarballs on the release page are just a copy of this repository with no source, links, or build files included.  

Please release source necessary to rebuild a binary release.  Releases of of the OS image for the CB1 board are currently violating the GPL.

thantik, 2022-08-13:
This is based off of the images of the OrangePi 2.  They haven't made any binary changes to anything, and thusly are not bound by the GPL.  Modifications to their distributions are all just added scripts, and you should contact OrangePi about their releases and any potential GPL violations.  I understand the rabid FOSS evangelism but your ire is misplaced here.

pendor, 2022-08-14:
GPL obligations aren't attached to modification.  GPL is a distribution license.  By distributing GPL code, BTT is required to also distribute source code.  If in fact they haven't modified anything, a *link* to the relevant source code is generally considered sufficient, but BTT hasn't done that either.  There's no indication that their release is based on the Orange Pi source tree.  Most importantly, there's no link or source for the CB1 device trees which are different from the Orange Pi to the point that releases built for Orange Pi aren't bootable on the CB1.

thantik, 2022-08-14:
GPL is a distribution license which applies to... CHANGES.  It stipulates that any CHANGES must be made publicly available.  And the CB1 is based off of the Orange Pi Zero 2.  The 1.0.1 hurican image even still contains the orange pi boot logo.

GPL v2 explicitly states:

1. You may copy and distribute verbatim copies of the Program's source code as you receive it, in any medium, provided that you conspicuously and appropriately publish on each copy an appropriate copyright notice and disclaimer of warranty; keep intact all the notices that refer to this License and to the absence of any warranty; and give any other recipients of the Program a copy of this License along with the Program. 

It is only upon **modification** that part 2 of this license applies.

The only violation here is their lack of a LICENSE file in the root of this repository.

shawnchr, 2022-08-14:
Go get a lawyer if you are so concerned???? They are very good about sharing information but they are starting to partner with Klipper and i think the info is changing constantly so until they get something more stable chill out always got to be some whiner.

pendor, 2022-08-14:
@thantik You said yourself the CB1 binary is based on the OrangPi firmware. That’s a modified version. They chose to release it which means they’re obligated to release source.  At a minimum they’ve modified the device tree which almost certainly came from U-Boot requiring release of any U-Boot modifications. They’ve also probably modified the kernel. 

There’s no exception in the GPL for “until they get something more stable.”  They released the binaries. They’re violating the copyright of those binaries until they also release source sufficient to rebuild those binaries. That obligation started when they released the binary. 

You can call it whining, but they’ve released hardware with software that doesn’t  work reliably in simple cases. Fixing that would be straight forward if source was available. Until then, this board is a doorstop.

thantik, 2022-08-14:
> @thantik You said yourself the CB1 binary

No, I have not - and you can check that yourself here.  No binaries have been changed by BTT.  Only configuration files.

I also never said anything about "until they get something more stable" that was someone else.

If you cannot follow a simple comment chain correctly, and attribute comments to their appropriate authors - maybe you also lack the authority to determine what is and isn't a GPL violation.

pendor, 2022-08-14:
I apologize for the mis-tag on the second part of my reply. GitHub issue comments aren’t an ideal medium for threaded conversation. 

BTT has plainly released binaries. They’re right here in this repository. The firmware images for the CB1 board are binaries that include and link to GPL covered code which makes the entire image covered by GPL.  

GPL requires release of complete corresponding source (CCS) capable of rebuilding any released binary. It doesn’t matter if they only modified config files or even if they made no modifications at all. There’s tons of precedent with LinkSys and other router manufactures. In those cases they released firmware which incorporated GPL code, and they were required to release source, scripts, config files, everything necessary to rebuild a functional firmware for those devices. 

From https://copyleft.org/guide/comprehensive-gpl-guidech6.html, Section 5.2.1:
`Indeed, CCS needs completely include not just that source which is directly translated by the compiler into object code, but other materials necessary to convert the source into equivalent binaries. Specifically, GPLv2 §3 requires that the source code include “meta-material” like scripts, interface definitions, and other material that is used to “control compilation and installation” of the binaries. In this manner, those further down the distribution chain are assured that they have the unabated freedom to build their own modified works from the sources provided.`

BTT isn’t in compliance with this requirement. Adding a copy of the GPL license text to this repo won’t bring the release into compliance. Compliance requires a full set of source, scripts, and config files that can reproduce the firmware images that are released in this repo.

thantik, 2022-08-14:
An image file of a running system does not constitute a binary release.  I already told you what distribution this is.  OrangePi already covers all of this in great detail, including all of their scripts and modifications.  All of which you ask for is located on their page.  A license file pointing to that page is all BTT needs to do to cover their asses here.  They have not "compiled" anything.  They've added some startup scripts (which I've stated already, are independent works) - and have run apt-get and kiauh (also separate works) on an Orange Pi Zero 2 image.  Go build it yourself.  It's all there.

pendor, 2022-08-14:
You’re wrong both about what constitutes a binary and also about the device tree not being modified. I’ve extracted the DTB from BTT’s images. They’re not the same as the OrangePi. The power management chip and SD card detect line are different in incompatible ways, probably among other things.

thantik, 2022-08-14:
Post your proof.

pendor, 2022-08-14:
Yes, OrangePi Zero 2. The one with the same H616 SoC as the CB1. The DTB’s don’t match. I’ll post my extract scripts when I’m back at a proper computer.

pendor, 2022-08-14:
[extract-dtb-sh.txt](https://github.com/bigtreetech/CB1/files/9334121/extract-dtb-sh.txt)
[01_dtbdump_,sun50iw9-dts.txt](https://github.com/bigtreetech/CB1/files/9334129/01_dtbdump_.sun50iw9-dts.txt)

Attached extract-dtb-sh.txt, (run as shell script) will extract DTB's from the Android kernel image, either of a mounted SD card imaged w/ BTT's firmware or from the decompressed IMG file itself.  Once decomplied, the DTB's are trivially easy to distinguish from those for the Orange Pi Zero 2, even in their mostly garbage decompiled back to source form.  Sample result extracted from the Debian flavored version BTT released is also included.

The OPiZ2 DTS I'm comparing to comes from the U-Boot source as according to the [Linux Sunxi site](https://linux-sunxi.org/Xunlong_Orange_Pi_Zero2), mainline U-Boot is the currently recommended repository.

Right off the bat, the model attributes are different "OrangePi Zero2" versus "sun50iw9", so either it's a modification or BTT's image isn't based off the current OPiZ2 release.

More significantly, the OPiZ2 DTS references an AXP305 PMIC whereas the CB1 decomplied DTS references an AXP1530.  The two boards use different power management chips, and using the DTS from the OPiZ2 isn't able to boot a functioning system on the CB1 hardware.  I've already tried that using the U-Boot source and user-mode from Armbian.  U-Boot dies complaining about not being able to set CPU & RAM voltages before the kernel loads.

Whatever BTT has released differs enough from the Orange Pi Zero 2 firmware that more than just a link to the Orange Pi firmware is needed to successfully build a functional firmware image for the CB1.

shawnchr, 2022-08-15:
What are your peoples problem???? They can't even get it to work right yet and you want source code for what????? So you can fix it??? Like I said before they just partnered with Kilipper this may not even stay with the current code you are wanting.

thantik, 2022-08-15:
Why are you comparing an Android kernel image to a Debian one?  They haven't released an Android build.  This is clearly a bad faith effort...

pendor, 2022-08-15:
@thantik Because BTT's released image contains an Android kernel bundle.  That's how they have their boot chain setup.  Maybe you'd care to do some actual research before slandering my work?

thantik, 2022-08-15:
Output of my cat /etc/*-release

BOARD=orangepizero2
BOARD_NAME="Hurakan"
BOARDFAMILY=sun50iw9
BUILD_REPOSITORY_URL=
BUILD_REPOSITORY_COMMIT=
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=
VERSION=3.0.4
LINUXFAMILY=sun50iw9
ARCH=arm64
IMAGE_TYPE=user-built
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
BRANCH=current
BOARD=orangepizero2
BOARD_NAME="Hurakan"
BOARDFAMILY=sun50iw9
BUILD_REPOSITORY_URL=
BUILD_REPOSITORY_COMMIT=
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=
VERSION=3.0.4
LINUXFAMILY=sun50iw9
ARCH=arm64
IMAGE_TYPE=user-built
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
IMAGE_UUID=2c153f1e-276e-40f2-800a-9fec288ff8bc
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

thantik, 2022-08-15:
Likewise - https://linux-sunxi.org/Allwinner_SoC_Family 

sun50iw9 (0x1823) | sun50iw9p1 | H616, H313 - is clearly the H616 identifier.

thantik, 2022-08-15:
This is an Orange Pi Zero 2.  Build tools are already available for it here: https://github.com/orangepi-xunlong/orangepi-build

thantik, 2022-08-15:
You posted a single dump.  You didn't post a diff of the dumps between the OrangePi Zero 2 build of Debian, and the build of Debian that BTT released.

pendor, 2022-08-15:
The diff is a useless mess because it's comparing a decompiled binary to source.  That's one of the reasons I want BTT to release their source...  

I posted the extracted DTB from BTT's image above, as well as the commands to extract it yourself if you'd like.  Here's a link to the stock Orange Pi Zero 2 DTB from the U-Boot tree which you can compare the dump to: https://github.com/u-boot/u-boot/blob/master/arch/arm/dts/sun50i-h616-orangepi-zero2.dts

You can't diff them directly, but you can find the same logical blocks within them.  I mentioned some of those differences above.  Based on my testing trying to get mainline U-Boot to run on this board, the primary issues are that the SD card "card present" line is different / non-functional on the CB1 (which can be worked around by adding "broken-cd" to the sdmmc block) and the power management chip is an AXP1530 on the CB1 instead of the AXP305 on the Orange.  That difference means none of the controllable voltage regulators are programmed properly, so U-Boot won't go out of its initial minimal boot mode and can't hand off to a full Linux kernel.

The OPiZ2 DTS has the AXP305 reference at line 111 whereas the extracted decompiled one I pulled from BTT's image has references to an AXP1530 starting at lines 2736 and 4441 as well as a reference to a axp806 at 4449, but no AXP305 to be found anywhere.  Attempting to boot the CB1 with the DTS from the OPiZ2 fails with errors about being unable to set voltages for the CPU and DRAM.  I don't have the full output from that attempt handy, but it's a moot point to my request for source.

shogowarrior, 2022-08-17:
GPL or not, IMO, it's helpful to have source code and more info for peace of mind

for e.g: 
* I had to search the repo to to figure out username/password 🤦 .
* `mirrors.tuna.tsinghua.edu.cn` 😮 
  ```sh
  [biqu@CB1 ~] cat /etc/apt/sources.list
  deb https://mirrors.tuna.tsinghua.edu.cn/debian bullseye main
  deb http://deb.debian.org/debian bullseye main contrib
  
  ```