bigtreetech/CB1 #11: BTT SKR Pico isn't detected until manually reset

pendor, 2022-07-31:
Each time the system boots, an attached SKR Pico isn't detected until ther reset button on the Pico is pressed.

**Immediately after Linux Reboot, before Pico reset button pressed**
```
pendor@v01:~$ lsusb
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 008: ID 1d50:614e OpenMoko, Inc. stm32f042x6
Bus 001 Device 007: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pendor@v01:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.170 (root@ubuntu) (gcc version 5.3.1 20160412 (Linaro GCC 5.3-2016.05) ) #415 SMP PREEMPT Fri Jul 29 18:02:16 PDT 2022
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] cma: Reserved 16 MiB at 0x000000007f000000
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   DMA zone: 4096 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 262144 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 22 pages/cpu @ffffffc03eef2000 s49792 r8192 d32128 u90112
[    0.000000] pcpu-alloc: s49792 r8192 d32128 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: root=/dev/mmcblk0p4 rootwait rw rootfstype=ext4 console=tty1 console=ttyS0,115200 loglevel=6 ubootpart=4e70dec2-01
[    0.000000] log_buf_len individual max cpu contribution: 131072 bytes
[    0.000000] log_buf_len total cpu_extra contributions: 393216 bytes
[    0.000000] log_buf_len min size: 131072 bytes
[    0.000000] log_buf_len: 524288 bytes
[    0.000000] early log buf free: 129496(98%)
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 972944K/1048576K available (12030K kernel code, 1166K rwdata, 4256K rodata, 2432K init, 692K bss, 59248K reserved, 16384K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008c40000   ( 12032 KB)
[    0.000000]     .rodata : 0xffffff8008c40000 - 0xffffff8009070000   (  4288 KB)
[    0.000000]       .init : 0xffffff8009070000 - 0xffffff80092d0000   (  2432 KB)
[    0.000000]       .data : 0xffffff80092d0000 - 0xffffff80093f3808   (  1167 KB)
[    0.000000]        .bss : 0xffffff80093f3808 - 0xffffff80094a08e8   (   693 KB)
[    0.000000]     fixed   : 0xffffffbefe7fb000 - 0xffffffbefec00000   (  4116 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf01000000   (    16 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc040000000   (  1024 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 24.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000005] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.002394] Console: colour dummy device 320x480
[    0.002516] console [tty1] enabled
[    0.002533] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002543] pid_max: default: 32768 minimum: 301
[    0.002636] Security Framework initialized
[... 805 lines cut ...]
Bus 001 Device 007: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 009: ID 1d50:614f OpenMoko, Inc. rp2040
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pendor@v01:~$ dmesg
>Snipped from listing above<
[   67.970137] usb 1-1.1: new full-speed USB device number 9 using sunxi-ehci
[   68.100471] usb 1-1.1: New USB device found, idVendor=1d50, idProduct=614f
[   68.100485] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   68.100493] usb 1-1.1: Product: rp2040
[   68.100500] usb 1-1.1: Manufacturer: Klipper
[   68.100507] usb 1-1.1: SerialNumber: 45503571270AA338
[   68.104716] cdc_acm 1-1.1:1.0: ttyACM1: USB ACM device
[   68.106638] snddaudio snddaudio0: codec: snd-soc-dummy, codec_dai: snd-soc-dummy-dai.
[   68.106669] snddaudio snddaudio0: ASoC: CPU DAI (null) not registered
[   68.112725] snddaudio snddaudio0: snd_soc_register_card failed
[   68.119668] sunxi-codec-machine sndcodec: ASoC: CODEC DAI sun50iw9-codec not registered
[   68.127969] sunxi-codec-machine sndcodec: snd_soc_register_card failed -517
[   68.135522] snddaudio snddaudio0: codec: snd-soc-dummy, codec_dai: snd-soc-dummy-dai.
[   68.135549] snddaudio snddaudio0: ASoC: CPU DAI (null) not registered
[   68.141617] snddaudio snddaudio0: snd_soc_register_card failed
pendor@v01:~$
```

This is repeatable 100%.  Pico should be detected at startup without requring manual reset.

FusedForm, 2025-03-24:
Hello, just solving this same issue.
It seems to be solved adding a delay in the HW_init function
Delay_ms(1500) in here:

  Delay_init(); 
  Delay_ms(1500); //  new delay

It seems that the screen needs a delay for giving time to the board to boot, then connect.

dmach, 2025-05-23:
The same issue here.
BTT SKR Pico was connecting successfully via usb in the can mode when there were no steppers connected.
When I connected the steppers, it disappeared after starting the printer.
Resetting BTT SKR Pico also helps in my case.