bigtreetech/CB1 #12: config.txt not found - cannot write user settings

rainerschulte, 2022-08-01:
I have written the image with Pi Imager and changed Settings for WiFi and languages to german regions.The Pi has now booted up and I see LAN LED's lighting and in the back near the WiFi Antenna flashing a green LED on the Adapter.

When flashing any Pi-Image to the CB1 nothing is booting, lichtning, flashing... except for one green LED on the CB1 which I beleive is the signal for some power on the CB1.

Can I copy a config.txt from an existing Pi to the SD-Card?

djscrubb, 2022-08-01:
no you have to edit system.cfg for wifi settings

rainerschulte, 2022-08-01:
So I did...  Double checked. Hardware defective?

oliver-eifler, 2022-08-01:
"When flashing any Pi-Image to the CB1 nothing is booting"
CB1 is not a PI and it doesn't work with any image for PI
You have to use the image from BTT CB1_Debian11_bullseye_minimal_kernel4.9_20220730.img 
https://github.com/bigtreetech/CB1/releases
You could write it with Raspberry PI Imager (or BalenaEtcher) to a SD (min. 16GB size)
After writing you'll find a FAT32 boot partition on SD. You have to edit the system.cfg which is on that partition with a editor capable of reading/writing linux type text files (on windows don't use notepad/wordpad, best is to use vs-code, atom, notepad++ or proton) to change your wifi settings. than boot CB1 with that sd ... may take some time for the first boot when setting up whole system.

rainerschulte, 2022-08-01:
The thing with RasPi I already understood and flashed the CB1_Debian you mentioned above with Balena Etcher. First I deleted all stuff on my SD-Card to have a fresh one. After flashing the CB1 Image, I inserted the SD into the Adapter and powered it up to get a FAT32 Partition with system.cfg where I changed the WiFI Settings according to my Network. 
Again, I cannot acces CM1 via WiFi even after waiting 15 Mins.

Aleks-MO, 2022-08-02:
@oliver-eifler  @bigtreetech   I used different SD cards (8-32 GB)... not on one there is no system.cfg, Windows 7 system cannot identify SD cards... how can I start WiFi via SSH?

oliver-eifler, 2022-08-02:
@rainerschulte Seemed to be the correct way ... What type of sd card did you use?. Some CB1 or images don't work with <= 8GB cards (was a long way to trouibleshoot).

@Aleks-MO: if you don't see the boot partition (even window 7 could display it as it is a FAT partition) something is wrong with your Image Software, SD Drive or SD. 

WiFi gets started only if there is no LAN connection (Network cable). If you could SSH into your CB1 via LAN you could use a tool like FileZilla or WinSCP on your Windows PC to browse and edit files on your CB1/Linux system.

rainerschulte, 2022-08-02:
@oliver-eifler I use a Sandisk 32 GB. I hooked it up with LAN and was able to configure remote Desktop. Once established, I got a Message that Wifi is restricted by policy. I'm asked for root password several times but it was denied. I tried _biqu_ and later I tried to change the PW via SSH. How can I change the policy for WiFi and what is the correct root-Password?

rainerschulte, 2022-08-02:
@Aleks-MO Mine was created during the very first boot in CB1. So at the beginning I had a EXT (Linux) Filesytem on the whole SD. 
After I inserted the Card the first time in CB1, all other stuff was created.

oliver-eifler, 2022-08-02:
@rainerschulte user should be biqu and its password should be biqu ... don't know the desktop but you could also change boot/system.cfg from shell/gui and configure your wifi settings and then reboot your cb1 with disconnected lan cable to connect to your wifi

oliver-eifler, 2022-08-02:
@Aleks-MO please explain what you have done and if the CB1 refuses to boot or if its only reject WiFi but you could connect via LAN

rainerschulte, 2022-08-02:
> @rainerschulte user should be biqu and its password should be biqu ... don't know the desktop but you could also change boot/system.cfg from shell/gui and configure your wifi settings and then reboot your cb1 with disconnected lan cable to connect to your wifi

nope... nothing changed.

oliver-eifler, 2022-08-02:
@rainerschulte as the btt image is based on debian for oran pi zero2 maybe the default password 'orangepi' will still work
Could you SSH into your CB1 via LAN?

rainerschulte, 2022-08-02:
I have written a brand new image without remote desktop and first commented out the LAN in system.cfg. Had no connection to WiFi. So I uncommented it again and connected to LAN and I was able to SSH. I have an IP and can connect by putty or MobaTerm. That's why I'm now affraid having a Hardware issue with WiFi Module on CB1.

Aleks-MO, 2022-08-02:
@oliver-eifler  I'll start from the beginning, so that it's clear what I got... I wrote the image to the SD card, there is no system.cfg, the Windows 7 system cannot detect SD cards. But I still installed the SD card on CB1 and downloaded, installed OCTOPRINT, communicated via MobaXterm_Personal and LAN.  After I turned off the LAN, but the WiFi did not work. Now I'm trying to enable it via LAN via SSH connection in MobaXterm_Personal by writing a corrected version of system.cfg to the BOOT folder.

Aleks-MO, 2022-08-02:
@rainerschulte  I have a similar problem, they also write about this problem on Facebook.... but for some reason it is solved differently for everyone....(((((

Magic2991, 2022-08-21:
hallo leute, ich bin neu. leider habe ich die selben Probleme. win10 erkennt keine partionen auf der 64gb karte. das erstellen der config schlägt fehl, weil PI Imager keine partion zum schreiben findet. lan geht nicht, muss ihn über wlan einrichten

rainerschulte, 2022-08-21:
Bei mir wurde die FAT32 Partition von Linux erstellt, wenn ich die Karte aus dem Rechner in den Pi Adapter gesteckt und gebootet habe. --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

closed 2022-09-04 by rainerschulte