bigtreetech/CB1 #13: Ok so I am having an issue My SSID on my wireless has a space in it and I have tried modify the /mnt/udisk/system.cfg with an escape character  WIFI_SSID="xx\ xxxxxxx" but I still getting Error: invalid extra argument 'Wireless'. Any help would be appreciated this work on the rasberry pis no issue even with no escape character.

shawnchr, 2022-08-02:
Ok so I am having an issue My SSID on my wireless has a space in it and I have tried modify the /mnt/udisk/system.cfg with an escape character  WIFI_SSID="xx\ xxxxxxx" but I still getting Error: invalid extra argument 'Wireless'. Any help would be appreciated this work on the raspberry pis no issue even with no escape character.

So it seems you script that is parsing the file is no taking into account space in the SSID which is a huge issue as I do not feel like changing 75 device to make the CB1 work. I could create a separate network but that defeat the purpose of a guest network.

_Originally posted by @shawnchr in https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3309774_

shawnchr, 2022-08-02:
So I was able to modify /etc/wifi_config.sh with the follwing and it will now parse the SSID properly with the space.

function read_wifi_config()
{
   IFS=''
   while read LINE
   do

But now I get now  I get the following

[biqu@CB1 ~]$ /etc/wifi_config.sh 
==>wifi config: ssid:My Wireless passwd:xxxxxxxxx
Error: No network with SSID 'My Wireless' found.

shawnchr, 2022-08-02:
So by adding IFS='' above to wifi_config.sh it will fix the space in the name issue but I had to run the the following commands
[biqu@CB1 ~]$ sudo nmcli radio wifi on
[biqu@CB1 ~]$ sudo nmcli dev status   
DEVICE         TYPE      STATE         CONNECTION         
eth1           ethernet  connected     Wired connection 2 
eth0           ethernet  disconnected  --                 
wlan0          wifi      disconnected  --                 
wlan1          wifi      disconnected  --                 
p2p-dev-wlan0  wifi-p2p  disconnected  --                 
p2p-dev-wlan1  wifi-p2p  disconnected  --                 
gre0           iptunnel  unmanaged     --                 
gretap0        iptunnel  unmanaged     --                 
ip6tnl0        iptunnel  unmanaged     --                 
sit0           iptunnel  unmanaged     --                 
tunl0          iptunnel  unmanaged     --                 
lo             loopback  unmanaged     --                 
ip_vti0        vti       unmanaged     -- 

[biqu@CB1 ~]$ sudo nmcli --ask dev wifi connect 'My Wireless'
Password: ************
Device 'wlan0' successfully activated with '20d9f6b1-46be-4657-9443-d47cddf169f5'.
[biqu@CB1 ~]$ sudo nmcli dev status
DEVICE         TYPE      STATE         CONNECTION         
eth1           ethernet  connected     Wired connection 2 
wlan0          wifi      connected     My Wireless        
eth0           ethernet  disconnected  --                 
wlan1          wifi      disconnected  --                 
p2p-dev-wlan0  wifi-p2p  disconnected  --                 
p2p-dev-wlan1  wifi-p2p  disconnected  --                 
gre0           iptunnel  unmanaged     --                 
gretap0        iptunnel  unmanaged     --                 
ip6tnl0        iptunnel  unmanaged     --                 
sit0           iptunnel  unmanaged     --                 
tunl0          iptunnel  unmanaged     --                 
lo             loopback  unmanaged     --                 
ip_vti0        vti       unmanaged     -- 

It would be nice to get this working with out manual intervention.

bigtreetech, 2022-08-03:
Thank you for your feedback. We will test and confirm it. and verify it with your solution. Thank you very much for your solution

shawnchr, 2022-08-04:
So had to reload the image from scratch because I broke ssh. After the fresh SD card reload I booted via ethernet made the IFS='' to the wifi_config.sh and then modified the system.cfg with my wireless credentials and this worked as expected. 

I still saw an issue on first boot I never got a network connection I had to kill the power and boot up again to even get an ethernet connection.