bigtreetech/CB1 #14: Provide Pinout for the module

StoneColdCrazy, 2022-08-03:
In the current version it is not apparent to what extent the CB1 is compatible with a CM4. Many functions are not possible (as an example the CSI port). By providing the pinout and the associated functions, technical incompatibilities can be derived more quickly. I have not been able to find any information on this so far.

With the provision, many questions would explain themselves.

emesix, 2022-08-03:
How do you know that the SPI bus is not connected to the "Header" ?
We are complaining that the linux distrobution is missing SPI support. If that support is working then we can check the SPI wiring to the header.

StoneColdCrazy, 2022-08-04:
I don't know that SPI has not been released, but as I said, there are no pinout descriptions that could help enormously. There are simply too many differences between CB1 and CM4.

Examples are among others the above mentioned PCIE on the CM4, since the H616 does not give this out of the chip, this will probably not be available. However, there is quite a bit the other way around. For example, the H616 has multiple USB outputs, but this may not be mapped on the header.

My point with the pinout is that this board is a good alternative to a CM4 in several areas aside from 3D printing. However, it should also be clear which pins go to the header here. Also this would help enormously to create own distributions, if the hardware would be known among other things by schematics.

looxonline, 2022-08-21:
A pinout should be coming soon.

ghent360, 2022-09-02:
Yes please pinout and more hardware details. I see there are some debug UART pads on the CB1, please document.

emersonsc, 2022-09-26:
> A pinout should be coming soon.

Its been over a month... where is it? Why hasnt Biqu/BTT released how THEY got input shaper working, since they claim this combo is compatible. I have a Manta 8 with a CB1 and unless i use the orange pi image and someone elses walkthrough I cant actually use an adxl and input shaper.

looxonline, 2022-09-26:
> Its been over a month... where is it? Why hasnt Biqu/BTT released how THEY got input shaper working, since they claim this combo is compatible. I have a Manta 8 with a CB1 and unless i use the [...]

Input shaper just works as long as you install the extra packages needed (as per the klipper guide) and use python 3. There is no, zilch, zero, nada secret sauce. Just specify the pins in the config and go...

emersonsc, 2022-09-27:
> Input shaper just works as long as you install the extra packages needed (as per the klipper guide) and use python 3. There is no, zilch, zero, nada secret sauce. Just specify the pins in the config [...]

Again... Where is the pin out? It was supposed to be provided. What's the pin out using a CB1?

emersonsc, 2022-09-27:
Also, where is the documentation on how to set up the adxl on the Manta and cb1 board? SPI header on the MCU or GPIO pins for the CB1? If so what's the pin out for the CB1 GPIO pins?

Michelinu, 2022-10-02:
After 6 weeks the GPIO PINOUT for the CB1 is still a mistery for BTT/Biqu. The CB1 is currently useless for whom needs GPIO pins.

I did some research and discovered that in the M4P there's a PINOUT for the CB1, here it is.
[image]

You have to take the name after the underscore when using the CB1. Then use this table https://wiki.pine64.org/wiki/SOPine_GPIO_Alternate_Functions_Table to get the `GPIO<number>`.
I tried with some "PC" pins and it seems to work

emersonsc, 2022-10-02:
> After 6 weeks the GPIO PINOUT for the CB1 is still a mistery for BTT/Biqu. The CB1 is currently useless for whom needs GPIO pins.
> 
> I did some research and discovered that in the M4P user manual there's a PINOUT for the CB1, here it is. [image]
> 
> You have to take the name after the underscore for the PIN you need. Then use this table https://wiki.pine64.org/wiki/SOPine_GPIO_Alternate_Functions_Table to get the `GPIO<number>`. I tried with some "PC" pins and it seems to work

It's sad no one here from BTT can bother to comment with something so simple. They designed the d*** board why is it so hard for them to release the information so we can utilize what they made and we paid for?

I also love the crickets on the adxl. We keep getting told it works... But no one wants to say HOW. No explanation, no walk through, no documentation. Just "it does work"

Michelinu, 2022-10-02:
> It's sad no one here from BTT can bother to comment with something so simple. They designed the d*** board why is it so hard for them to release the information so we can utilize what they made and [...]

As far as concerns ADXL345 I tried to configure it by using the informations provided by stas2z here: https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3540300

I'm still waiting to receive my ADXL so I can't test it to confirm that it works. I'm using latest version of CB1 official firmware (2.2.0) and I don't get any error causing Klipper to not boot. Theorically by just wiring it correctly it should work.

emersonsc, 2022-10-02:
> As far as concerns ADXL345 I tried to configure it by using the informations provided by stas2z here: [#9 (comment)](https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3540300) Since his wiring scheme looks a bit messy, I made a new one [image]
> 
> I'm still waiting to receive my ADXL so I can't test it to confirm that it works. I'm using latest version of CB1 official firmware (2.2.0) and I don't get any error causing Klipper to not boot. Theorically by wiring it correctly, it should work

I saw those steps, but it requires the use of an image from another manufacturer. BTT claims it works with theirs, but had provided nothing as far as HOW. 

Did you follow everyone one of those steps to make it work with the BTT image? If you had to change things, can you provide your steps?

Michelinu, 2022-10-02:
> I saw those steps, but it requires the use of an image from another manufacturer. BTT claims it works with theirs, but had provided nothing as far as HOW.
>
> Did you follow everyone one of those steps [...]

He stated that it got it working by using his custom OrangePi image for CB1 - he doesn't exclude that it might work also with the latest official images from BTT. If you flash the "Klipper" image all the required libraries, NumPy and the linux MCU host are already installed. If you are using the "minimal" image you have to install them by following the Klipper documentation: https://www.klipper3d.org/Measuring_Resonances.html
 
Since the `raspi-config` command is not available, I added the following entries in my `/boot/BoardEnv.txt` to enable SPI:
```
overlays=spi-spidev
param_spidev_spi_bus=1
param_spidev_spi_cs=1
param_spidev_max_freq=1000000
```
Running `ls -1 /dev/spidev*` I get `/dev/spidev1.1` as output - so I imagine that it  did the trick.
Trying using the `GPIO74` as `OUTPUT_PIN` and testing it with a multimeter, I have the confirmation that pin n.15 in the 40-pin header (that normally is `GPIO22` - the cyan wire in my wiring diagram) is `GPIO74` in CB1.

Here's my current `adxl345.cfg` file (that you can #include in your `printer.cfg` or paste the content directly in that):
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:gpiochip0/gpio74
spi_bus: spidev1.1

[resonance_tester]
accel_chip: adxl345
probe_points:
  100, 100, 20
```

Running `sudo cat /sys/kernel/debug/gpio` displays that `gpio-74` is used by Klipper.

As said before, I still didn't get my ADXL345 so I can't confirm it's really working

Borillion, 2022-10-22:
> GPIO Access Denied #44

Thank you this is very helpful :)

Borillion, 2022-10-24:
[image]

Michelinu, 2022-10-24:
> [image]

"I can not get it for you" -> "We can make a pinout diagram/scheme/table since we "developed" it. We did our job and you already bought it, so we are not interested at using our time to solve your problem/doubt. Do it by yourself, if you can".

Since the company is not interested at publishing further documentation about the CB1, I created a thread that regroup all the info already published by some users and the ones I discovered: https://github.com/bigtreetech/CB1/discussions/47

looxonline, 2022-10-24:
> "I can not get it for you" -> "We can make a pinout diagram/scheme/table since we "developed" it. We did our job and you already bought it, so we are not interested at using our time to solve your [...]

I have been discussing this with them. I am not sure why they have not released it yet. They should have. I have all of the data and will try to put something together soon. In the meanwhile, if anyone has a specific question about a pin then they can ask me.

Borillion, 2022-10-27:
> I have been discussing this with them. I am not sure why they have not released it yet. They should have. I have all of the data and will try to put something together soon. In the meanwhile, if [...]

Not much to go by but started mapping out some of whatI can find from the datasheets of the different carrier boards into a csv. 

https://github.com/Borillion/CB1/blob/main/CM4toCB1_GPIOMapping.csv

emersonsc, 2022-11-20:
> A pinout should be coming soon.

Where's the pinout? Its been over 3 months.

mutlubasdas, 2022-12-05:
> He stated that it got it working by using his custom OrangePi image for CB1 - he doesn't exclude that it might work also with the latest official images from BTT. If you flash the "Klipper" image [...]

Hey,
Were you able to test it and see if this wiring and config works?

frutabruta, 2023-01-04:
The pinout has been released:  [Link](https://github.com/bigtreetech/CB1/tree/master/Hardware)

Borillion, 2023-01-04:
@frutabruta thank you. @bigtreetech this is so very much appreciated, and was a long time coming. <3

looxonline, 2023-01-05:
> @frutabruta thank you. @bigtreetech this is so very much appreciated, and was a long time coming. <3

Schematic is also up on the official github. You'll see quicker turnaround on the pinout in the future. Schematic will have a delay of a few months to prevent competitors from copying design points.

looxonline, 2023-01-05:
> @looxonline Consider adjusting the policy for releasing pinouts.
> 
> If the team released pages 9 and 10 documenting A1-100 and B101-200, it would have made the module much easier to use by the community without disclosing any internal wiring "secrets".
> 
> By hiding the entire schematic, the community was not able to understand how to implement projects outside the source code that your team distributed. This really limited the usefulness of your product for months.
> 
> Limiting usefulness is not a benefit to anyone.

Absolutely agreed. I only started working for BTT three days ago but this is something that I am going to be like a dog with a bone. I will hang onto it with each release to make sure that it happens. The full pinout will be released in a well crafted manual for each subsequent SBC product from now on...and there are some awesome SBC products in the pipeline ;)

looxonline, 2023-01-05:
> Congratulations on your new position!
> 
> As a recommendation for the other boards, clarify which pins are for the MCU vs the CB1.
> 
> Exposed Names for user-side pins without knowing which chips they go to make schematics misleading. Here's an example: [bigtreetech/Manta-M4P#8](https://github.com/bigtreetech/Manta-M4P/issues/8)

Also a good point. This is done well on the 40 pin header but not elsewhere in the sch. Will work on that.

Jckf, 2023-01-13:
If anyone else ends up here looking for the CB1 GPIO numbers; [here's a spreadsheet I put together](https://docs.google.com/spreadsheets/d/1ayO3U5eCA0b5ZhrTTsDDRVX2oHeO4XbG9rNFywYXwNc/edit?usp=sharing) using the information @Michelinu posted above.