bigtreetech/CB1 #15: Spi or klipperscreen

djscrubb, 2022-08-08:
Has anybody been able to get
A) klippersceeen actually working either otg or via WiFi on an android device
B) inputshaper via Adxl345 via spi
Thanks

pl4nkt0n, 2022-08-23:
No, still not working. Tried to use the CB1 with the pi pad 5. But it look like the CB1 doesnt support it!!
Total fail: you have a display with ports for CB1 (or CM4) but cannot use this bigtreetech display with a bigtreetech pi - total useless crap and waste of money...

MugenMicko, 2022-08-31:
Unable to get a Waveshare 4.3 DSI screen to function on it either.  Does the CB1 support anything bar a BTT TFT??

acwalter, 2022-09-12:
CB1 does not support DSI/CSI output. You have to use HDMI for display and USB for camera and touchscreen.

When trying to enable adxl, I get MCU error during config: Unable to open spi device

frutabruta, 2022-09-12:
> Has anybody been able to get A) klippersceeen actually working either otg or via WiFi on an android device B) inputshaper via Adxl345 via spi Thanks

The ADXL via SPI can be enabled with different image (tutorial included):
https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3540300