bigtreetech/CB1 #16: CB1 WiFi

rmilyard, 2022-08-09:
So I wrote the CB1 image.

In the system.cfg file I put in my Wifi info. I did the 2.4ghz since not sure if 5 supported. I am looking for IP and not finding it on network. On the Manta board there is a soild RED LED near the CPU and green LED blinking near the CB1.

shawnchr, 2022-08-09:
After it booted on my network I had to remove power and plug it back in again for it to actually connect.

Does you wifi name have space in it?

rmilyard, 2022-08-09:
Nope no spaces.

shawnchr, 2022-08-10:
Did you try removing the power and power it back on after a fresh install I always had to do this to pull and IP.

rmilyard, 2022-08-10:
I did remove power.

rmilyard, 2022-08-10:
Noticed the ethernet LEDs are on with nothing plugged in.  The Orange and Green LEDs on solid.

shawnchr, 2022-08-10:
The ethernet LEDS do stay on currently when even connected to wifi on mine as well hopefully that get corrected in the future. BUt I am still connected to wifi.

rainerschulte, 2022-08-10:
I can connect to LAN, but WiFi doesn't connect. I have a simple SSID whithout space.

thantik, 2022-08-13:
There are 2 distributions for this board - I'm just passing along my experiences with this garbage.  V1.1.0 Debian 11 just kernel panics for me, I can never get it working.  Never connects to wifi.  I even connected a monitor, keyboard, mouse, and did

`sudo nmcli dev wifi list` -- will list my spaces-laden SSID...
`sudo nmcli dev wifi connect "3D Print Orlando" password mypassword` says that the SSID cannot be found though clearly listed otherwise.

I had to go back to the V1.0.1 Hurrican image, flash it, load it up in a virtual machine, remove the symlink /etc/systemd/system/multi-user.wants/hostapd.service - and then boot it on hardware with a keyboard and mouse attached.  Let it stay booted for a while - it might reboot once in order to expand the filesystem image.  Upon its reboot it still gave me issues and never dropped me to a login prompt - it would continue to bitch about ping, and not being able to connect to some random AP; but a CTRL+ALT+F2 managed to get me to the login prompt.  From there I stripped /etc/rc.local of the line which runs the script in /home/biqu/scripts/init.sh (leave the exit 0 line) -- and was able to run the nmcli command to connect to my wifi ap without trouble.  You can then strip all the scripts from the /home/biqu/scripts directory - they are garbage and biqu isn't doing any of this correctly.

You can then issue

`sudo nmcli dev set wlan0 autoconnect yes` - and upon reboot, the currently connected wifi should come up without the hassle of logging into it every time.  From here, you should have the ability to clean up the image.  You'll want to issue

`git config --global --unset https.proxy
git config --global --unset http.proxy`

In order to clear git of its proxy settings for a local machine they were using - this will allow git to connect to the internet and properly update things.  Then run:

`git clone https://github.com/th33xitus/kiauh.git
./kiauh/kiauh.sh`

And I would suggest using that helper script from there.

You'll also want to 

`sudo nano /etc/apt/sources.list' and change your sources from the chinese servers to normal debian servers

`
deb http://deb.debian.org/debian bullseye main contrib non-free
deb http://deb.debian.org/debian-security/ bullseye-security main contrib non-free
deb http://deb.debian.org/debian bullseye-updates main contrib non-free
deb http://deb.debian.org/debian bullseye-backports main contrib non-free
`

This at least got me to the point where I could start using the board, connecting to wifi, not using keyboard/mouse/hdmi, and simply powering on the board and SSHing into it.

RPK3D, 2022-08-14:
I can't connect the WLAN, installed the latest version for the CB1 and entered everything properly into the cfg file. But nothing happens, my SSID contains spaces

shawnchr, 2022-08-15:
@RPK3D if you can reimage with a fresh load and connect it via ethernet before you modify the system.cfg.
you can edit the /etc/wifi_config.sh like the following you need to add the IFS='' to get spaces to work. I submitted this fix to BTT.
function read_wifi_config()
{
  **IFS=''**
   while read LINE
   do
   
 The modify the system.cfg by editing /mnt/udisk/system.cfg with your creds and it will work.

AnAwesomePossum, 2022-08-17:
I'm in the same boat, edited the config file with my ssid and pw, none of which have spaces. I can connect with ethernet and When I do sudo nmcli dev wifi list my wifi is listed, and when I do sudo nmcli dev status it says my wlan0 is connecting configuring but never makes it to wifi. Also seems to drop connection from ethernet every couple minutes or so. This is on the newest CB1 lite image but I've used every one from v1 to now with no results.

shawnchr, 2022-08-17:
AnAwesomePos yep Thats where I am and I am fighting to get it replaced with tech support I have a big post on the facebook group might want to luke know you are seeing the same issue maybe BTT will get of there ass and realize they have an issue.
Shawn "That Guy" Chronister in teh Big Tree Tech Offcial Group

AnAwesomePossum, 2022-08-17:
> AnAwesomePos yep Thats where I am and I am fighting to get it replaced with tech support I have a big post on the facebook group might want to luke know you are seeing the same issue maybe BTT will [...]

Will do! It sucks that this thing just refuses to work.

BeigeAnimal, 2022-08-18:
I could be wrong, but it seems like the antenna just has pitiful range. 
I got it to connect to my router almost perfectly fine when I was setting it up at my desk, my router is on my desk.
but when I actually tried using it in my print it won't connect or would connect for two seconds before disconnecting.
I wrapped a wire around the antenna, and pointed my router's antenna directly to my printer and it seems to work.
it could be coincidence tho. but if other people could try testing the CB1 while in the same room as the router to confirm that be nice. keep mind there is a jumper for usb power on the Manta M4/8P.

Note: I did do what thantik did a few posts up.

shawnchr, 2022-08-18:
It not the antenna I was right next to my router and it will not connect

BeigeAnimal, 2022-08-18:
> It not the antenna I was right next to my router and it will not connect

Did you do what Thantik did? it could be a combination of things.

rainerschulte, 2022-08-18:
I tried Thantik's suggestion and found, that the old image doesn't boot my CB1. Then I came back to a newer version and got a connect via LAN. Anyway... I tried all "garbage" to delete and again I came out without any wireless connection. All requests returned "Creds not provided". I will now flash a fluidd on it and use the Board wired network. I hope it provides the possibility to use a USB-Cam on my Switchwire. That's weird...

RPK3D, 2022-08-18:
So @shawnchr's hotfix helped a bit to make it work.
I still had to change my wireless connection via "sudo nmtui". The CB1 tried to dial in using WPA3 encryption, after I selected WPA and WPA2 it worked.

shawnchr, 2022-08-18:
Thantik went back to an unsupported image. No I did not because I have successfully configured the current image and it works if you follow my steps unless there is hardware issue. I am fighting with support they keep trying to say it a software issue and I disagree.

I have this running with an M4p + CB1 and this was my M8p+CB1 I tested because I saw all the issues so I have one thats fine and one that not both built the exact same way.

rainerschulte, 2022-08-18:
@RPK3D That's the solution! WPA3 is not supported! I switched to WPA2 and I'm online!!!! HOORAY!!!

AnAwesomePossum, 2022-08-22:
guys, I powered my board via usb and suddenly my wifi is working! BTT mentioned something about it in their facebook group when trying to troubleshoot another problem so I gave it a whirl. Of course, now I need to know how to power it from the a PSU and keep the wifi working but I suppose it's a step.

kshensley71, 2022-08-24:
> @RPK3D if you can reimage with a fresh load and connect it via ethernet You have to wait for all lights to stop and unplug and power back up before you modify the system.cfg. you can edit the /etc/wifi_config.sh like the following you need to add the IFS='' to get spaces to work. I submitted this fix to BTT. function read_wifi_config() { **IFS=''** while read LINE do
> 
> Then modify the system.cfg by editing /mnt/udisk/system.cfg with your creds and it will work.

I bought 2 of these and neither of them worked (Wifi) the new release OS just kernel panics both boards, but the older release loads and works from LAN but not WLAN, I have no spaces in my wifi, but THIS worked for me! BUT my system.cfg was at /mnt/tfcard/system.cfg   I changed the system,cfg on the sd card and verified it was changed, but when I did a sudo nano /etc/tfcard/system.cfg it had a biqu wifi and password even though when I did a >nmcli d it showed MY wifi and in the state it said connecting (but it wasnt) so there is definitely something messed up in the wifi settings. Fortunately, mine is working now thanks to this, but that was 3 hours wasted on such a simple thing.

basha318i, 2022-08-24:
> So @shawnchr's hotfix helped a bit to make it work.
> I still had to change my wireless connection via "sudo nmtui". The CB1 tried to dial in using WPA3 encryption, after I selected WPA and WPA2 it worked.

this is what actually worked for me, thanks a lot!

shawnchr, 2022-08-25:
> > So @shawnchr's hotfix helped a bit to make it work.
> > I still had to change my wireless connection via "sudo nmtui". The CB1 tried to dial in using WPA3 encryption, after I selected WPA and WPA2 it worked.
> 
> this is what actually worked for me, thanks a lot!

Your welcome I was trying to find a fix for everyone now if BTT would get the new image out so I can load and prove I have a bad board it would be great. This board should never have been released into the wild and no support is even worse. This iwill turn off alot of customers epecially when Raspi cm4's come back I am dumping this crap.

basha318i, 2022-08-25:
> Your welcome I was trying to find a fix for everyone now if BTT would get the new image out so I can load and prove I have a bad board it would be great. This board should never have been released [...]

I too believe I have a bad board; the WiFi is extremely slow like when I was downloading Klipper it was at 235KB/s I understand the WiFi module on the CB1 is not so great & maxes at 100Mb/s but at least achieve 80% of that! Also, my PWR LED (the one next to ACT) doesn’t come on at all, I tried the Hurakan & I didn’t have the ACT working; then the latest which I got ACT, but no PWR LES at all; although it’s up & I’m able to access it.

thiagoshima, 2022-09-02:
You guys can test this, use nmtui to connect to the wifi, using wlan0, and then remove any ethernet connections, tested on the lastest image

AnAwesomePossum, 2022-09-02:
They sent out a new image to a few people. Wifi works over PSU power now. The screen isn't blurry. Ethernet does not work. They did say it was a pre release though.

kildiyarov, 2022-09-02:
> They sent out a new image to a few people. Wifi works over PSU power now. The screen isn't blurry. Ethernet does not work. They did say it was a pre release though.

Can you please share this image?

ma1069, 2022-09-02:
Did it solve the issue for you? I still see the same problem as before: wifi working over USB but not over PSU

AnAwesomePossum, 2022-09-02:
Yes and no. Wifi is working over PSU ethernet doesn't. It also appears that wifi will drop or become unresponsive. Nmcli will give me an ip but can't ssh into it after so long. I'll post a link to the image in a few.

billru05, 2022-09-03:
The new image is on the images page for the CB1

kshensley71, 2022-09-03:
Unfortunately my second Pi4 adapter board appears to be defective. I tried the new image using the adapter and a CB1 and nothing happens. Lights come on but no video and no wifi or network. I tried another CB1 and the same thing, so I tried one of the older images and nothing. It looks like my PI4 adapter board is no good. My first one is mounted in a printer that I am using so I can't remove it to test the new image on it right now, and I'm traveling for the next week. When I get home I'll pull it and test the image on it. Sent from my Galaxy

TxBillbr, 2022-09-18:
I have a usb wireless dongle (supported by buster drivers) on one and a wired usb dongle on another cb1. Not using the BTT networking allows me to move forward a bit. At least I am operational.

Gavrik13, 2023-11-02:
Hello.

Can you tell me how to beat this on the latest releases?

Dersonje, 2024-04-24:
I had to disable WPA 3 on my router to get 2.3.4 on the cb1 to connect to wifi. Tried everything else I reasonably could and that is the only thing that worked

dayrecker, 2024-04-26:
Highly recommend NOT following BTT instructions to edit system.cfg prior to booting the E3EZ. Instead, leave system.cfg alone connect via ethernet and then connect to your wifi via nmcli. Also, I switched my router to WPA/WP2 from WPA2/WPA3 for the initial connection (not sure if that was necessary), but no problems connecting again after turning network back to WPA2/WPA3.

Puding07, 2024-10-13:
In my case the problem was that my password included a $ sign which cut off the rest of the password...
Just to give you a basic udnerstanding what's happening is after boot an init script `/boot/scripts/btt_init.sh` runs `/boot/scripts/connect_wifi.sh` which will get the Wifi credentials from `system.cfg` file. The script generates a `.nmconnection` keyfile in `/etc/NetworkManager/system-connections/`. The generated file will be used to connect you to your wifi network.

Useful commands:
- Check your connections with: `sudo nmcli c s`
- Reload keyfiles: `sudo systemctl restart NetworkManager`

How I resolved the issue?
1. Before first boot I edited the recommended `system.cfg` file with my Wifi credentials.
2. After first boot I edited the init script with `sudo nano /boot/scripts/btt_init.sh` and commented out this line `./connect_wifi.sh &`
3. Edited the credentials in my keyfile: `sudo vim /etc/NetworkManager/system-connections/SSID_NAME.nmconnection` 
4. Reloaded keyfiles for Network Manager: `sudo systemctl restart NetworkManager`
5. Connected to Wifi: `sudo nmcli con up SSID_NAME`

After reboot it should automatically connect so you don't have to repeat these steps.

Additionaly if you want you can add hostname to your device if you like to have verything organised.
Add this line to you keyfile under ipv4 settings: dhcp-hostname=BTT-CB1.

Hope this helps ypu guys. 🙏
Cheers!

djuttien, 2024-12-10:
i finally fixed it by adding the wifirouter's defailt gateway ip adress to the system.cfg

win R cmd
ipconfig reads the default gateway
then
edit system.cfg
add (at the top somewhere)
ip=192.168.XX.XX )defaultgateway you just found
also
uncomment# host=BTT-CB1 (or a hostname of your preference
and
add wifi credentials as shown by Biqu