bigtreetech/CB1 #17: Manta M8P/CB1 not booting

jrsphoto, 2022-08-10:
I recently received my Manta board with the CB1 module and I am unable to get the board to function.  I have tried the following things:

- Downloaded your custom image from https://github.com/bigtreetech/CB1/releases/tag/V1.1.0
- Wrote that image to the 32mb SanDisk SD card
- Configured wifi setting to my WIFI
- Tried with Ethernet but the Ethernet connection never became active. At no time to I have lights on the Ethernet jack
- Tried several SanDisk SD cards from 8gb to 64gb
- Powering the board with 12v
- Powering the board with USB-C  and V-USB jumpered. (12v supply disconnected)

At no time did the CB1 board seem to boot, and that's after waiting 10 minutes of waiting, also no IP address is being issued by my DHCP server.

I did notice that the 4 USB-C connector shield solder points were not soldered and the USB-C connector is loose but I don't intend on using that if I can avoid it.

I do get a red light on the Manta board next to the reset buttons, and a green LED2 on the CB1 board, but no other indication of functionality. 

I feel like I'm missing something stupid but nothing I can see that's obvious.  I tried sending this to the btt support email but it bitched about "Mail content denied" so I figured I'd try here.

bigtreetech, 2022-08-11:
[image]
Hello, please confirm whether all the dial switches are set to off

jrsphoto, 2022-08-11:
They are indeed all to the left.

jrsphoto, 2022-08-11:
I did some further testing today and got slightly different results.   All of my microSD cards are Sandisk, 64gb, 32gb and one 16gb cards.  I did some digging around in my 3d printer parts drawers and found another Sandisk 16GB card and wrote [the image](https://github.com/bigtreetech/CB1/releases/tag/V1.1.0) to the card, edited the wifi info and tried another boot-up.  This time, I'm getting the red power light on the Manta board, the green LED on the CB1, and a blinking ACK led on the lower right edge of the Manta board.  The board is still never getting an IP address, either via WIFI or Ethernet, in fact, the Ethernet port never comes active but occasionally the right led of the Ethernet port blinks.

I also tested everything with a different Lamda 24v PSU and got the same results.

jrsphoto, 2022-08-12:
@bigtreetech - At this point I'm pretty sure there is something wrong with my board..  How should I proceed?

TheRevenantStar, 2022-08-13:
I am also experiencing the same issue with the M8P + CB1 + 16GB SD card.

TheRevenantStar, 2022-08-13:
Strangely after unbolting the heat sink, it's now booting... might have just been seated wrong.

jrsphoto, 2022-08-13:
You actually got the small screws for the CB1?  I didn't receive them so my CB1 is currently just held on by the connectors, which are very tight.  I did re-seated the CB1 board several times over the past few days, which may have been why I started to get the blinking ACK led.  I don't know what else to try.  I may try mess with it again today but I'm starting to think I may just be screwed (or maybe screwless?)  Quick question @TheRevenantStar, where did your board ship from.  I seem to remember the address on the package as some place in Illinois but my friend ordered his and it was coming from China.

TheRevenantStar, 2022-08-13:
The Illinois address is the US warehouse, anytime I order, it travels via that warehouse.
The bolts are M2, I got them with the heat sink.
Did you try to re-write the card using the pi imager and allow it to finish verifying?
Also if you're not on Windows, did you eject the card correctly?

jrsphoto, 2022-08-14:
I am on windows and always let the images verify before I use them.  I’ve probably been through that process about 50 times at this point.  For some reason I didn’t get a heat sink.  Thought I ordered one but I must not have.

jrsphoto, 2022-08-16:
@bigtreetech ? hello?

AnAwesomePossum, 2022-08-16:
I'm in the same boat. One time I managed to get additional lights to light up but still no IP. Using different cards made no difference for me.

AnAwesomePossum, 2022-08-16:
Well, I'm dumb, i was putting the sd card in the wrong slot. Getting lights like before but still no wifi even after editing the config file.

jrsphoto, 2022-08-17:
The documents were never clear about which SD Card slot to use so I always tried both.  It became clear pretty quickly that the card on the long bottom edge of the board was the only one that I could get the ACK led to light up.   Completely dead here though.  Thought about trying to get a Raspberry Pi CM4 module but its more than I paid for this pile of crap so I think I'll cut my losses now and go a different path.

shawnchr, 2022-08-17:
if you connect via ethernet can you run the ses commands
[biqu@CB1 ~]$ sudo nmcli dev status
DEVICE         TYPE      STATE                     CONNECTION         
eth1           ethernet  connected                 Wired connection 2 
wlan0          wifi      connecting (configuring)  My Wireless        
eth0           ethernet  disconnected              --                 
wlan1          wifi      disconnected              --                 
p2p-dev-wlan0  wifi-p2p  disconnected              --                 
p2p-dev-wlan1  wifi-p2p  disconnected              --                 
gre0           iptunnel  unmanaged                 --                 
gretap0        iptunnel  unmanaged                 --                 
ip6tnl0        iptunnel  unmanaged                 --                 
sit0           iptunnel  unmanaged                 --                 
tunl0          iptunnel  unmanaged                 --                 
lo             loopback  unmanaged                 --                 
ip_vti0        vti       unmanaged                 --                 
[biqu@CB1 ~]$ sudo nmcli dev wifi list
IN-USE  BSSID  SSID  MODE  CHAN  RATE  SIGNAL  BARS  SECURITY 

IN-USE  BSSID  SSID  MODE  CHAN  RATE  SIGNAL  BARS  SECURITY 

To see i the connection is get created

jrsphoto, 2022-08-17:
That would be great if it ever got an IP address.. but as mentioned in other posts, the wifi or Ethernet never get ip addresses.  In fact, the Ethernet port never goes active as far as I can tell.

jrsphoto, 2022-08-19:
At this point, I would recommend that anyone wanting to try out the Manta board with the CB1, should wait..  maybe forever, but at least until another rev of the board is released and they fix some pretty big issues with the board.  I REALLY REALLY wanted to like this board.  I wanted it for my Voron 2.4.  But I'm getting no response from support so I think that says it all.

AnAwesomePossum, 2022-08-19:
> At this point, I would recommend that anyone wanting to try out the Manta board with the CB1, should wait.. maybe forever, but at least until another rev of the board is released and they fix some [...]

100% agreed. I know being an early adopter comes with working out bugs and quarks but it should at least mostly work. I waited three weeks to get this from china when my octopus arrived next day from amazon. Maybe I'll just cut my losses, get another octopus and an orange pi.

shawnchr, 2022-08-19:
I agree @jrsphoto the fact they fail to support their product and dismiss anyone with a hardware issue because they found a kernel bug and think that going to fix everyone is short sighted and have lost me as a customer. I must admit I have an M4p+ CB1 and it is slick but lack of support ruins it.

AnAwesomePossum, 2022-08-19:
The cb1 will be releasing on US amazon on august 19th. I would never buy a new one and return the old one but it might be nice to have a spare 😉

koira, 2022-08-27:
I experience just the same as many others here. Manta M8P with CB1 doesn't boot form the sdcard, and with the CB1 detached the board is not being recognized as USB-device by a raspberry pi. looks like we have to wait for some serious updates.

jrsphoto, 2022-08-31:
I'm not sure an update will fix any of our issues.  I suspect they are faulty hardware.  Even the USB-C port on my Manta board was not properly soldered and came loose.

AnAwesomePossum, 2022-08-31:
> I'm not sure an update will fix any of our issues. I suspect they are faulty hardware. Even the USB-C port on my Manta board was not properly soldered and came loose.

I don't disagree with this. How unethical would it be to buy anew one off Amazon and return the defective one?

shawnchr, 2022-08-31:
About as unethical as its going to be when they tell us all they will not replace our bad ones because its been over 30 days.
These people are gerking our chains.

jrsphoto, 2022-08-31:
I like how you think.  

JS
John Scherer
***@***.***
619-806-7488

thiagoshima, 2022-09-02:
Having the same issue, at most I managed to install mainsail, but it only lasted one reboot and its totally broken now, what a headache from such a company!

yordanov-alexandar, 2023-03-07:
@all I had the exact same problem - CB1 wouldn't boot. At first all(1,2,3,4) my DIP switches from the image in the second comment were in ON position(the board came that way) and the CB1 board had no light at all. After finding this issue I moved switches 3 and 4 to OFF position and I got a green light on the CB1 board but still no boot and no activity on the ethernet port. After that I moved switches 1 and 2 to OFF position also and everything started working - board booted and there was activity on the ethernet port and also the activity light next to the CB1 board was blinking too. 
So if the image in the second comment gave you the impression that only switches 3 and 4 need to be in the OFF position then try switching 1 and 2 to OFF too.
If they are already OFF then it's probably not the same problem that I had. 
Just wanted to share this in case anyone (like me) had the impression(given by the highlighted point 5) that only switches 3 and 4 need to be OFF

fabio-s-franco, 2023-05-29:
So encouraging to see many people having the same issue and not a single word from @bigtreetech after a single screenshot from the manual which I am sure we all have already read. This lack of follow up commitment is just great and the manual very incomplete, without a single troubleshooting guide.
Well, I am going to post what I figured out, however, I am not sure what happens if the problem is slightly different, as I will detail below:

1 - I followed the same procedure. Went to 3/4, tec on off. I figured if I kept 3 and 4 on ON, it would re-run the boot process. But apparently this only happens once and I don't know how to rest if you, for example, used the wrong image, wifi pwd, etc.
2 - After going through the initial boot process, it turns out, it is a one time thing and I probably there is a way to start it again, but it is not documented anywhere.
3 - Then after putting them all to the left like in the manual (all off) , it booted and I was able to SSH into it. By the way, for those looking for another undocumented instruction: The SD card slot is the `SOC` not the MCU one. So if it is not working, just keep trying on the SOC otherwise you won't know what's wrong if you happen to have the SD CARD on the wrong slot.

There is a RESET and BOOT buttons in the board, but I don't know how to trigger them, if they can ever be used to restart this process.

So:
1 - Image to SD Card
2 - 3/4 switches to ON
3 - Wait blinks and so on, when there is nothing blinking anymore, you can turn power off the board (also not documented)
4 - Now, everything to the left and the card still sitting in the SOC slot
5 - SSH and continue to software installation instructions. Which is what I am about to go through at this exact moment, now that I succeeded SSH into it.

Given the experience I had so far, I am not too optimistic. May Gordon Moore (RIP) be with me now in this harsh time.

ChuiVanfleet, 2023-12-19:
I found that if I pressed down on one corner of the cb1, the green light would come on. After finally finding some bolts to fit into the four corner mounting holes, I applied power and it booted up. But I can now ssh in.

dawesy72, 2023-12-20:
Comforting to know I'm not alone. Tried all combinations of switches and as card slots a nothing. 
Only with the SD card in the bottom/ long side slot do I get the pi fan led, flashing act led and activity on the Ethernet port ( not showing on up scan though) if I put it in the short side as card slot I get nothing bar the led2 on the CB1 and the 3 red leds over by the power connectors. 
Will try out of the printer tomorrow on usb power.

HumbertoPerez0018, 2024-09-25:
Same issue but using the PI4B board. I wished I read this before buying front @bigtreetech, also I ordered a CB2 but never arrived so I'm happy to get a refund.

RPI alternatives are bad enough that there's some hacking around, but I can't get this board to work it's sad to see these problems are nothing new and there's no response from @bigtreetech

kmjr20, 2024-10-22:
I had the board working and tried to transfer to the cb2 emmc. As soon as i rebooted the device, nothing comes up on the screen or the angry IP. I have tried reinstalling several times with a new SD card and still nothhing boots up. All the lights on the board are flashing like its working, but it has no connection to the network or wifi anymore. I put the old CB1 back on, and it still id not boot.

Pneumanifest, 2024-10-22:
> I had the board working and tried to transfer to the cb2 emmc. As soon as i rebooted the device, nothing comes up on the screen or the angry IP. I have tried reinstalling several times with a new SD [...]

Did you unzip the image file before flashing it to the eMMc?

kmjr20, 2024-10-22:
Yes I did. I followed the directions to a T. I did not get the same screens as the instructions say. There was no choices to write to the Emmc or format using the Ext.  I got a lot of error messages from the RKDev tool and had to do it 1 file at a time. I put on the old CB1 to see if it was the CB2 causing the problem, but it does the same thing. I purchased other SD cards and they did not work either. I set the jumper to use USB power and the USboot o and RPI switch as directed, it no longer shows up in RKDev. How do I reflash the board so that it works again?

FLB-daniel, 2025-02-15:
> I found that if I pressed down on one corner of the cb1, the green light would come on. After finally finding some bolts to fit into the four corner mounting holes, I applied power and it booted up. But I can now ssh in.

This exactly fixed it for me.
I struggled for days getting it to boot, i got green light on CB1 and one red power led on manta board. Couple of times i got ack green light but never got to full boot.

After this post i tightened the screws more than i normally would feel comfortable with doing, an voiula, booted instantly and connected to WiFi all as i should!

valir, 2025-05-18:
Found this issue while havving issues booting my Manta M8P v2.0 for the first time. That was because it's not clear what SD-card slot should be used. In my case, I've successfully booted Bigtreetech image and also the Armbian image out of the SD-card labeld "SD-SOC".