bigtreetech/CB1 #18: CPU Warnings and Reboot (CB1 + Manta M8P)

TheRevenantStar, 2022-08-13:
I'm seeing warnings on the screen like the following and then some kind of crash dump with the message "Rebooting in 5 seconds..."
There are more warning than these listed... but I can't read them off this tiny screen fast enough.
`
WARNING: CPU: 0 PID: 3039 at sd_recv0xdc/0x374 [8189fs]
`
`
rt[8189fs]: probe of mmc1:001:1 failed with error -123
`
There was a warning something about "card claims to support voltages below defined range"

After some time, the system reboots on its own. There does not seem to be an exact pattern.
Additionally, I'm not getting a WiFi connection even though I defined one in the system.cfg file before plugging in the SD card.
I'm using the Manta M8P and the CB1

TheRevenantStar, 2022-08-13:
To add to my previous comment, I have tried to write the OS using the Pi Imager on both MacOS 12.5 and Windows 11

thantik, 2022-08-13:
This isn't a solution to the problem - but I just wanted to post this in an effort to help people get going for now until BIQU can tackle these issues in their official releases.

https://github.com/bigtreetech/CB1/issues/16#issuecomment-1214230225

frutabruta, 2022-08-15:
Have look at this post, it might be a hardware failure/bad soldering:
https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3396322

AnAwesomePossum, 2022-08-19:
Exact same thing. Errors just like yours and no wifi. I can see wireless networks by running nmcli dev wifi but theyre always just connecting. Made sure they're not set to wpa3 by running nmtui and still no dice. I can connect through ethernet without issue.

arrowcircle, 2022-08-28:
Same here. Both wlan0 and wlan1 stuck in connecting state. There is a chance it connects to wifi, but I cant find meaningful things here. Any news on how to fix this?