bigtreetech/CB1 #19: full specifications of `system.cfg` needed

vtta, 2022-08-17:
I need the ability to add custom kernel command line arguments. Currently there is no obvious way to do this. I want to know if such modifications could be achieved using `system.cfg`. So it would be easier if you can provide the full specifications of this config file. Or the better way is to just open source the whole source code, so that we can build our own distribution or kernel image.

shawnchr, 2022-08-17:
You cannot pass kernel params via this file system.cfg is used as a variable in the /etc/wifi_config.sh and it is call by the /etc/NMconnection script that starts NetworkManager that manages the network interfaces.

[biqu@Micron1 ~]$ cat /etc/wifi_config.sh 
#!/bin/bash

WIFI_CONFIG_SSID=(null null)
WIFI_CONFIG_PASSWORD=(null null)

INDEX=0

WIFI_CONFIG_FILE=/mnt/udisk/system.cfg

ghent360, 2022-09-02:
Kernel params are on the second partition of the sdcard image. Not sure what the format of the partition is, probably uboot something, but you can see the kernel params as clear text.

vtta, 2022-09-03:
Than

> Kernel params are on the second partition of the sdcard image. Not sure what the format of the partition is, probably uboot something, but you can see the kernel params as clear text.

Thanks for the info, I will give it a try 👍

VIXIVIXIV, 2022-09-18:
> > Kernel params are on the second partition of the sdcard image. Not sure what the format of the partition is, probably uboot something, but you can see the kernel params as clear text.
> 
> Thanks for the info, I will give it a try 👍

Do you have any further updates on passing kernel parameters? I haven't been able to get it to work either.

vtta, 2022-09-18:
No. I have already ditched the CB1 due to super bad WiFi card and lack of source code. Now I have switched to CM4.

closed 2022-09-18 by vtta