bigtreetech/CB1 #21: UART configuration

shogowarrior, 2022-08-18:
Hi All,

I needed help with configuring `UART0`. I see it as `/dev/ttyS0` but am unable to connect anything to it

shogowarrior, 2022-08-18:
Also what is the root passwd?

bkrengineer, 2022-08-18:
biqu

EchoHeim, 2022-11-08:
> Also what is the root passwd?

The password is the same as the username.
root user is root,and biqu is biqu

closed 2024-03-07 by bigtreetech

Escrich, 2024-03-22:
Something strange happens now, in the beginning was like you wrote:

The password is the same as the username.
root user is root,and biqu is biqu

Today, I was mounted the last image available, but of course biqu works, and I can change password without a problem, but, if you try to write 'su' at command line, the system ask you for a password, but the password is not biqu, and is not the one you changed for the common user, if you try to do ssh root@your_ip, again yo can connect with the machine, and again asks you for a password, and of course again, the password is not biqu, and is not the one you generated for the common user , have this been changed?