bigtreetech/CB1 #22: Change user name to pi

MDFPereira, 2022-08-22:
Hello.
Is there any possibility to use the normal user name, pi?
biqu is messing some permitions and it would be much easier to have it as pi.

shawnchr, 2022-08-22:
What permission? if you are referring to gcode that is because they forgot to change the permissions before compressing the image that have lost of issues that need fixed biqu user is not the problem.

TxBillbr, 2022-09-02:
Installing with kiauh correctly chooses the users and changes the scripts to use biqu user. The only permissions problem I found was the root-owned gcode directory (which one can remove, or change owners and perms.

ToyVo, 2022-09-14:
you could rename the user with these three commands
```sh
usermod -l NEWUSERNAME biqu
usermod -m NEWUSERNAME -d /home/NEWUSERNAME
groupmod -n NEWUSERNAME biqu
```

you should probably disable root login with `passwd -l root`

jscancella, 2023-01-03:
Making the user `pi` would greatly help with creating other images using customPiOS