bigtreetech/CB1 #23: Wifi works powed by USB but not PSU

AnAwesomePossum, 2022-08-23:
Using an m8p with cb1, connected to USB power the wifi connects and stays connected. When connected to PSU power wifi will see the networks but never connect. Ethernet when connected to PSU will drop connection ever few minutes.
On FB BTT asked about this exact scenario when troubleshooting someone else's problem and said they had a solution but didn't say what it was.

shawnchr, 2022-08-23:
Yeah thats the way I had been running all of mine. But the people at BTT seem to think making me wait for the stupid software will fix my issue. I am done with BTT and will spread the bad support far and wide.

tnap1979, 2022-08-25:
you aint the only one...
im still struggling, only thing that works is one image with ethernet,
wifi is shit... ive added caps, ive tried different images... kernels...
fuck this garbage

ma1069, 2022-09-02:
This issue still persists in the last image: in the release notes they specified it was meant to solve wifi issues, but I noticed no differences whatsoever, at least on my board :(

AnAwesomePossum, 2022-09-02:
So there's a secret squirrel image they've sent out to a few that has wifi working over PSU power. Ethernet doesn't work though and wifi seems to drop connection. I'll upload the file to a Google drive when I get home.

AndreOkihara, 2022-09-22:
I'm having the same problems with wifi not working with PSU. (M8P + CB1)
After a lot of tests I found something interesting, if I take all the stepper driver out, the wifi connects and is stable (for 1 hour of testing).
My assumption for now:
When power on by PSU, everything is power at the same time and the CB1 might get power fluctuations that messes with wifi connection.
I'm doing more tests for stability and soon put an oscilloscope on the power rails to verify my assumptions. I'll post the results here.

tnap1979, 2022-09-22:
> I'm having the same problems with wifi not working with PSU. (M8P + CB1) After a lot of tests I found something interesting, if I take all the stepper driver out, the wifi connects and is stable (for 1 hour of testing). My assumption for now: When power on by PSU, everything is power at the same time and the CB1 might get power fluctuations that messes with wifi connection. I'm doing more tests for stability and soon put an oscilloscope on the power rails to verify my assumptions. I'll post the results here.
> 
> Maybe a delay in process initialization fixes this issue.

The problem stems from hardware fault..
The LDO regulator is not good enough..
It needs to be replaced with a torex semiconductor sot23-5 3.3v LDO 750mah or higher...

AndreOkihara, 2022-09-22:
> 

You are absolutely right, I'm just probe the CB1 3.3V rail and see a drop to 1.4V 😞 
I'm gonna try remove the sot23 and put a external 3v3 regulator to test.

AndreOkihara, 2022-09-22:
#

> The problem stems from hardware fault.. The LDO regulator is not good enough.. It needs to be replaced with a torex semiconductor sot23-5 3.3v LDO 750mah or higher...

You are absolutely right, I'm just probe the CB1 3.3V rail and see a drop to 1.4V 😞
I'm gonna try remove the sot23 and put a external 3v3 regulator to test.

ADKMechETECH, 2022-10-10:
> The problem stems from hardware fault.. The LDO regulator is not good enough.. It needs to be replaced with a torex semiconductor sot23-5 3.3v LDO 750mah or higher...

Do you have a part number or other recommendation for specs? I'm only finding regulators with 2% regulation and wasn't sure if that was good enough.

unique311, 2022-11-05:
The LDO regulator that's in the revised version 2.2 CB1 has an ID H63MY.
Not finding it anywhere online.

tnap1979, 2022-11-06:
> The LDO regulator that's in the revised version 2.2 CB1 has an ID H63MY. Not finding it anywhere online.

https://smd.yooneed.one/

Try that site.. that's what I use to identify my smd components

tnap1979, 2022-11-07:
> The LDO regulator that's in the revised version 2.2 CB1 has an ID H63MY. Not finding it anywhere online.

H63	XC6220B331MR	Torex Semiconductor
SOT-25	Linear voltage regulator IC
LDO, 3.3V±1%, 1A, +CE, CL