bigtreetech/CB1 #24: SSH for CB1+pi4B adapter

davbcr, 2022-08-23:
I finally get to work and connect to wifi my CB1 on the Pi4B adapter but i cannot connect any screen because HDMI resolution is not accepted by them and i cannot go in via ssh because i don't know which the default user and password are.....
pi + raspberry is not working as well as cb1 + bigtreetech 
Can i write some file on SD Card configuring ssh for CB1? If yes what should i write into it? (sorry newby questions)

shawnchr, 2022-08-23:
User is Biqu and the password the same that was at least in the manual.

davbcr, 2022-08-23:
Which manual? I've got nothing in the box with it.

frutabruta, 2022-08-23:
> Which manual? I've got nothing in the box with it.

Those are available as PDF on GitHub of Manta M8P/Manta M4P/Pi4B adapter.

shawnchr, 2022-08-23:
[BIGTREETECH_M4P&CB1 User Manual.pdf](https://github.com/bigtreetech/Manta-M4P/blob/master/BIGTREETECH_M4P%26CB1%20User%20Manual.pdf)

davbcr, 2022-08-24:
Thank you so much. i got in.

closed 2022-08-24 by davbcr