bigtreetech/CB1 #25: PI4B + CB1 No MCU found ! USB seem not to work

Giorgiofossati, 2022-08-23:
Hello, i'm using PI4B + CB1. 
I installed KIAUH , Klipper and Moonraker without problem, wifi work perfectly, but i can't connect to the printer the CB1 can't see the MCU wich was previously flashed with klipper. I need to reflash the MCU with some particular USB setting?
[image]

shawnchr, 2022-08-23:
What ls /dev/serial/by-id/ show?
What MCu are you using?

Giorgiofossati, 2022-08-23:
@shawnchr [image]

MCu is an AtMega 2560

shawnchr, 2022-08-23:
OK I am not sure you can use anything but a BTT board. There is supposed to be new image coming maybe it will work then in the mean have tried recreating firmware on the PI4B+CB1 and reflashing MCU not sure it will work but might try it. Lots off issue with this board and software.

frutabruta, 2022-08-23:
> @shawnchr [image]
> 
> MCu is an AtMega 2560

More important is the type of UART->USB chip. For example CH340, PL2303,  CP2102.
The drivers just might not be included in the system.

Giorgiofossati, 2022-08-25:
@frutabruta lsusb say this, it seem it could see usb devices…
[image]

frutabruta, 2022-08-25:
> @frutabruta lsusb say this, it seem it could see usb devices… [image]

So CH340 it is. Try to install drivers using this tutorial: [CH340 install](https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/linux)

Giorgiofossati, 2022-08-30:
@frutabruta  hi and thanks for the replies, I have not been able to install the CH340 drivers. see the images
[image]

rainerschulte, 2022-09-04:
same to me!

shawnchr, 2022-09-04:
This is image
https://github.com/bigtreetech/CB1/issues/29#issuecomment-1236130800
You will need to use nmtui via a monitor to configure wlan0 to get wifi connected
the see if you can do your drivers.

rainerschulte, 2022-09-04:
> 

I don't have any HDMI Monitor left over. nmtui can also run from command line. This is how I changed my network settings until now. the large hook is WEP3 not compatible. l must switch back to WEP2.