bigtreetech/CB1 #26: Cannot find files mentioned in skr pico uart setup

RedSinkDirty, 2022-08-26:
Cannot find files mentioned in skr pico uart setup, It seems the boot folder is empty? are these files on a different partion?

these are the instructions I cannot follow from the skr pico setup doc:
If you use UART0 to communicate with raspberry pi, you need to modify the following files by inserting the SD card into the computer or by SSH command. And wiring reference [here](https://github.com/bigtreetech/SKR-Pico/blob/master/Klipper/README.md#raspberry-pi-is-powered-by-the-motherboard-5v-and-communicates-with-skr-pico-v10-via-uart)
Remove console=serial0,115200 in /boot/cmdline.txt
Add dtoverlay=pi3-miniuart-bt at the end of file /boot/config.txt
Modify the configuration of [mcu] in printer.cfg to serial: /dev/ttyAMA0 and enable restart_method: command by SSH
[mcu]
serial: /dev/ttyAMA0
restart_method: command

ateixeira07061992, 2022-09-25:
Having the same problem. I'm able to follow the USB guide but unable to do it via UART .

RIP-PloKoon, 2022-11-09:
Any updates on this?

ateixeira07061992, 2022-11-10:
[image]
 I'm now waiting

RIP-PloKoon, 2022-11-12:
Got it working! Used new image (v2.2.1), edited configs per readme, made new klipper.uf2 and re-flashed the BTT SKR Pico.

MatteoAvalle, 2023-01-06:
How did you do that @RIP-PloKoon ? There still are no cmdline.txt nor config.txt inside the boot partition, on 2.2.1

RIP-PloKoon, 2023-01-06:
I just followed the readme in this repo. [Here](https://github.com/bigtreetech/CB1#overlays-settings), first bullet in the overlay settings. Then as stated above, “made new klipper.uf2 and re-flashed the BTT SKR Pico.”

Ntufao, 2023-03-20:
@RIP-PloKoon which image did you use? can you please send me the link? I've used several versions and I'm not getting it. Can you send the link to the image you used?

Thank you very much

RIP-PloKoon, 2023-04-03:
I just used the current at the time (v2.2.1)