bigtreetech/CB1 #29: v2.1.2 Wifi Issues

AnAwesomePossum, 2022-09-03:
So I can get wifi connected with the 2.1.2 image over psu but it doesn't last. For me when it stops working the entire system stops working and requires a hard reset but cutting power. Sudo reboot now and shutdown now just send it into an endless splashscreen. I have not been able to determine how long wifi stays connected but it's less than 10 minutes. In my picture below I took it as soon as ssh became unresponsive so in that case it was 3 minutes.

AnAwesomePossum, 2022-09-03:
[image]

AnAwesomePossum, 2022-09-03:
Update: 30 minutes later and now it refuses to connect to wifi at all.

billru05, 2022-09-03:
Yes, same here also the image will break if you try to update or write anything to it, the only way I can get it to connect again once it disconnects and refuses to connect is to reformat the SD card and reload the image then it will repeat again. I am puzzled how BTT keeps posting broken files.

RcTomcat1, 2022-09-03:
Can't confirm wifi or update issues.
Sources.list contains a bunch of chinese update servers, image seems to be synced with those servers (mirrors.tuna.tsinghua.edu.cn). I added the regular bullseye sources to my system and just did an apt-get update & upgrade. Also 0 Updates available. 
mirrors.tuna.tsinghua.edu.cn is also on the official debian mirrors list, so all good.

Wifi and ssh connected for over 20 minutes now, did the update process without issues and still working fine.

Killajoedotcom, 2022-09-03:
Absolutely cannot get the CB1+M8p to connect at all. The M8P guide doesn't seem to help at all, and doesn't correct the problem. Anyone get this working?

billru05, 2022-09-03:
Hello RcTomcat1, I changed the Chinese source list to the below and still same issues,
deb http://deb.debian.org/debian bullseye main contrib non-free
deb-src http://deb.debian.org/debian bullseye main contrib non-free

deb http://deb.debian.org/debian bullseye-updates main contrib non-free
deb-src http://deb.debian.org/debian bullseye-updates main contrib non-free

deb http://deb.debian.org/debian bullseye-backports main contrib non-free
deb-src http://deb.debian.org/debian bullseye-backports main contrib non-free

deb http://security.debian.org/debian-security/ bullseye-security main contrib non-free
deb-src http://security.debian.org/debian-security/ bullseye-security main contrib non-free

billru05, 2022-09-03:
Here is what I just discovered, when I power on the CB19 (with the latest image) I get an IP address about 10 secs before the green LED by the power port starts flashing, as soon as the light starts flashing the CB1 is dropping out

RcTomcat1, 2022-09-03:
@billru05 i used the following:
deb http://deb.debian.org/debian bullseye main
deb-src http://deb.debian.org/debian bullseye main

deb http://deb.debian.org/debian-security/ bullseye-security main
deb-src http://deb.debian.org/debian-security/ bullseye-security main

deb http://deb.debian.org/debian bullseye-updates main
deb-src http://deb.debian.org/debian bullseye-updates main

But as the chinese mirror is just that: a mirror there should be no need to update the sources.list
The flashing green LED signals just a working system as far as I know.
What does the router log and the system log of the CB1 say?

shawnchr, 2022-09-03:
> Update: 30 minutes later and now it refuses to connect to wifi at all.

There is something wrong with these boards I have my original 1 that works great no matter what I put on it but the last 2 I received do not work connect for few minutes and stop.
I can run this image on my board that works and get all the features[
](https://github.com/stas2z/orangepi-build-cb1/releases/tag/0.1)

Killajoedotcom, 2022-09-03:
> There is something wrong with these boards I have my original 1 that works great no matter what I put on it, but the last 2 I received do not work connect for few minutes and stop. I can run this [...]

Which one of the 4 are you using?

frutabruta, 2022-09-03:
> There is something wrong with these boards I have my original 1 that works great no matter what I put on it, but the last 2 I received do not work connect for few minutes and stop. I can run this [...]

I also think it's a hardware issue on some of the boards. Both of my CB1s (ordered the day of launch) have stable wifi with the 2.1.2 image, without any problems. The same image does not work for other people.

cellarboy, 2022-09-03:
> Which one of the 4 are you using?

I've tried both images this morning. The Bullseye image kernel panics once it boots and the Bookworm image boots, but Wifi appears completely non functional.

cellarboy, 2022-09-03:
> So I can get wifi connected with the 2.1.2 image over psu but it doesn't last. For me when it stops working the entire system stops working and requires a hard reset but cutting power. Sudo reboot [...]

This is just appalling - how could they possibly have thought this works well enough to release?

shawnchr, 2022-09-03:
> Which one of the 4 are you using?

On the board that works I am using the bullseye minimal image and it works great. This image or any other image fails to work properly on my later boards. The key with this image is it loads all there proper kernel modules and if you watch it boot up with a monitor connected it has no red errors.

shawnchr, 2022-09-03:
> I've tried both images this morning. The Bullseye image kernel panics once it boots and the Bookworm image boots, but Wifi appears completely non functional.

I only have one this works on. I am positive they have a manufacturing issue and there 2.1.2 image still is lacking the Image I reposted above is great if your have a good board.

billru05, 2022-09-03:
I brought six at one time (I ordered them 1 month ago) I just got them last week and I have played with two of them and I get the same results on any image or tweaking I load, they may get wifi for about 2 mins then they drop wifi, I can not get it back until I. clean & reformat the SD card reload image and again it may or may not get wifi but if it dose get wifi if drops within 2 - 3 mins. I can reboot, I can let it set still dose not get the wifi back.

TxBillbr, 2022-09-03:
I have seen other folks have problem with the wifi due to BTT leaving WPA3 set in the system config. That is probably worth checking. Both of my boards (recently purchased from KB-3D.com in the US) work ok with my wifi.

shawnchr, 2022-09-03:
> I brought six at one time (I ordered them 1 month ago) I just got them last week and I have played with two of them and I get the same results on any image or tweaking I load, they may get wifi for [...]

Yeah this is a batch or manufacturing issue I am sure of it. No way this can be software because like I said the one good board I have can run all of the images with wifi. 2 later ones do not work.

TxBillbr, 2022-09-03:
I have installed 2.1.2 on one of the two boards received from KB-3D.com (US Vendor) and wifi has been running well for about 4hrs thus far. I have been playing with gpiod and sunxi-pio and those seem to be working correctly as well.

billru05, 2022-09-03:
Hello shawnchr , I just loaded the image  from the link you posted (https://github.com/stas2z/orangepi-build-cb1/releases/), and both LED on board are solid green and I have no wifi, so there is differently something of a mystery on this board

shawnchr, 2022-09-03:
> Hello shawnchr , I just loaded the image from the link you posted (https://github.com/stas2z/orangepi-build-cb1/releases/), and both LED on board are solid green and I have no wifi, so there is differently something of a mystery on this board

With that image you need to go into nmtui and setup wlan0 it not configured automagically and the lights do stay solid on the board I have working so it could still crappy design.

billru05, 2022-09-03:
Hello shawnchr, there are no lan lights on board so I do not think I can connect, I will try but the Lan looks dead on board

shawnchr, 2022-09-03:
> Hello shawnchr, there are no lan lights on board so I do not think I can connect, I will try but the Lan looks dead on board

No no networking of any kind initially you will need a monitor and keyboard

TxBillbr, 2022-09-04:
Has anyone found a way to read what version the hardware  is with software?  Also the scripts used at startup log cleartext credentials for SSID/password to /etc/scripts/wifi.log. Take a look at the mess in /etc/scripts. init.sh sets the timezone to HK :-( 

logs should be in /var/logs (especially system related ones). No credentials should ever be logged. No assumption should be made of timezone or wifi country code.

AnAwesomePossum, 2022-09-04:
So using the orangepi bullseye is getting me farther than btt did. The wifi is working but still seems to want to drop every now and then. While ssh putty will hang while trying to install something, in this case it was klipper. When I sudo reboot it hangs on "A stop job is running for" and lists three things (it's hard to see on the monitor). I'm going to try the other image because why not.

shawnchr, 2022-09-04:
> So using the orangepi bullseye is getting me farther than btt did. The wifi is working but still seems to want to drop every now and then. While ssh putty will hang while trying to install [...]

So this the stop job takes forever to stop but it will eventually stop and reboot. The issue with the wifi dropping I am telling this is a hardware issue. the one board I have the works never drops so something happened with the later boards.

TxBillbr, 2022-09-04:
> Has anyone found a way to read what version the hardware is with software? Also the scripts used at startup log cleartext credentials for SSID/password to /etc/scripts/wifi.log. Take a look at the [...]

As a follow-on, I am running fine on wifi on one board while the other absolutely refused. I finally commented out /etc/rc.local and installed a USB networking dongle and now both hardware and wifi work on that board.  IMHO they are shooting themselves in the foot trying to do the dynamic flop/flip of ap  and  wireless via the rc.local (which kills wpa_supplicant and does all sorts of non-standard shinanigans. )   I also set my locale, my wifi country, and timezone. Now it feels almost workable!

AnAwesomePossum, 2022-09-04:
Okay, so using the orangepi bullseye WITH a wifi dongle I am connected via PSU power, been able to install klipper, moonraker, fluidd, etc and so far it's been over an hour and I'm still connected via ssh! I'm not sure if this would work on the btt 2.1.2 image they just released but it is, so far, definitely working on this one. I shouldn't have to use an external piece of hardware to get their board working but considering they'll probably not make this right, I'll take what I can get. I'll update in the morning if anything changes

rainerschulte, 2022-09-04:
I tried both images now and I cannot connect wheter it's LAN or WiFi. This is the only stability I notice Nothing is working! Even after hours waiting for a sign of life. What I see is the Power is on and a blinking LED below WiFi module. The PiAdapter is powered via PSU or directly on my Laptop. Ethernet was connected in my office and powered by PSU or Router. LED's on LAN were dark. No lights or blinking... Great Job Guys! The same results as promises from german government. Thumb's up!!!

aodxxl, 2022-09-04:
So i installed 2.1.2 this morning and everything went great. used the minimal image , connected to the CB1 via SSH right away and started to configure everything. I updated everything, installed KIAUH, then Klipper, moonraker and mainsail and connected to the web interface. Modified the config file for my printer and decided to take a break. At this point, the system was on for over an hour.
A few hours later i turn the system back and and surprise... its not connecting to wifi. In the meantime i did found a workaround by reediting the config with the guest network. Now i am connected to the guest network and got access to the board again.
Aparently using a new SSID gives you access once again. might not be a hardware issue after all and just bad software

TxBillbr, 2022-09-04:
Apparently they broke the network driver in 2.1.2. If you comment out everything in /etc/rc.local, their scripts will stop killing the wifi every 5 seconds (and will stop the blinking led).  For those of you not familiar, rc.local gets executed every time you login. From my limited testing  the "check interval" seems to cause severe severe thrashing of the network stack as it kills wpa_supplicant and reconfigures the wifi for AP mode. In 2.1.2 they changed the value from 30 sec to 5 sec. It appears that the first reconfigure takes a while (as the system comes up)  and second invocation starts before the first one finishes.  I've been running two boards successfully for 24 hours now by dropping the pile of scripts from rc.local that instigate the rtl crashes.  

I've also found that nmtui is a fairly useful tool in this exercise. It is a text-based network manager client.  

The one good thing I've found is the script /etc/scripts/pwr_status, which it what blinks the power LED. So at a minimum we know that gpio229 is the LED! LOL

shawnchr, 2022-09-04:
> So i installed 2.1.2 this morning and everything went great. used the minimal image , connected to the CB1 via SSH right away and started to configure everything. I updated everything, installed [...]

This is not bad software this hardware. The orange pi image has the same problem on the boards with the issue and I can tell for a fact that orangepi image on the one board I have that works with no matter which image I put on it. That one board I also get all the function your are supposed to with this board it has no hardware issue.

I do believe there software is still broken too but not for wifi.

aodxxl, 2022-09-04:
> > So i installed 2.1.2 this morning and everything went great. used the minimal image , connected to the CB1 via SSH right away and started to configure everything. I updated everything, installed KIAUH, then Klipper, moonraker and mainsail and connected to the web interface. Modified the config file for my printer and decided to take a break. At this point, the system was on for over an hour. A few hours later i turn the system back and and surprise... its not connecting to wifi. In the meantime i did found a workaround by reediting the config with the guest network. Now i am connected to the guest network and got access to the board again. Aparently using a new SSID gives you access once again. might not be a hardware issue after all and just bad software
> 
> This is not bad software this is a hardware problem. The orange pi image has the same problem on the boards with the issue and I can tell for a fact that orangepi image on the one board I have that works with no matter which image I put on it. That one board I also get all the function your are supposed to with this board it has no hardware issue.
> 
> I do believe there software is still broken too but not for wifi.

Its software. Most likely in some early batches they used a diffrent power management Chip or some other components that is now incompatible with the software  and the devs working on these issues use the early batches to test things instead of the latest units. 
Honestly it could also be a defective batch of chips as i dont think there are a lot of early adopters.

looxonline, 2022-09-04:
> Okay, so using the orangepi bullseye WITH a wifi dongle I am connected via PSU power, been able to install klipper, moonraker, fluidd, etc and so far it's been over an hour and I'm still connected [...]

Are you on facebook? If you are, contact me on messenger. I am an admin on their group. Look for my profile pic.

AnAwesomePossum, 2022-09-04:
> Are you on facebook? If you are, contact me on messenger. I am an admin on their group. Look for my profile pic.

Yes I am, I made the same post on FB. I just tagged you.

looxonline, 2022-09-04:
> This is just appalling - how could they possibly have thought this works well enough to release?

Hit me up on FB messenger if you have FB. I will try to get you in touch with some peeps at BTT.

looxonline, 2022-09-04:
> Apparently they broke the network driver in 2.1.2. If you comment out everything in /etc/rc.local, their scripts will stop killing the wifi every 5 seconds (and will stop the blinking led). For those of you not familiar, rc.local gets executed every time you login. From my limited testing the "check interval" in the system.cfg seems to cause severe severe thrashing of the network stack as it kills wpa_supplicant and reconfigures the wifi for AP mode. In 2.1.2 they changed the value from 30 sec to 5 sec. It appears that the first reconfigure takes a while (as the system comes up) and second invocation starts before the first one finishes. I've been running two boards successfully for 24 hours now by dropping the pile of scripts from rc.local that instigate the rtl crashes. A usb network dongle works around these issues (wireless or wired).
> 
> I've also found that nmtui is a fairly useful tool in this exercise. It is a text-based network manager client.
> 
> The one good thing I've found is the script /etc/scripts/pwr_status, which it what blinks the power LED. So at a minimum we know that gpio229 is the LED! LOL

Interesting. The first connection attempt was supposed to take 5s and if that fails then all after that should take 30s. At least that is the algorithm that we discussed. I need to check the scripts in the latest image though.

AnAwesomePossum, 2022-09-04:
> Apparently they broke the network driver in 2.1.2. If you comment out everything in /etc/rc.local, their scripts will stop killing the wifi every 5 seconds (and will stop the blinking led). For [...]

Tried commenting out everything in rc.local (there was only one script) but still the same results. Did you do anything else?

billru05, 2022-09-04:
Can anyone confirm the dates on the CB1 that are working vs the ones that are not, this is my dates and nothing works:
CB1 (1020000395 BIGTREETECH CB1_V2.1 1G D/C 2022.7.27
PI4B 1030000103 BIGTREETECH PI4B Adapter V1.0 D/C 2022.8.1
I have six of the above, I have played with two of the six and I get the same results also the dates are the same on all boxes

shawnchr, 2022-09-04:
> Can anyone confirm the dates on the CB1 that are working vs the ones that are not, this is my dates and nothing works: CB1 (1020000395 BIGTREETECH CB1_V2.1 1G D/C 2022.7.27 PI4B 1030000103 [...]

I cannot tell you my pi adapter date no longer have the box but my pi adapter works great
I also no longer have the box for the one that works I looked and no longer see it. But will keep looking

BUt I can tell you the date on 2 boards aren't working 
2022.7.27
2022.7.30

billru05, 2022-09-04:
So it appears something before 2022.7.30 changed or bad componets

frutabruta, 2022-09-04:
> Can anyone confirm the dates on the CB1 that are working vs the ones that are not, this is my dates and nothing works: CB1 (1020000395 BIGTREETECH CB1_V2.1 1G D/C 2022.7.27 PI4B 1030000103 [...]

Both working after resoldering one wi-fi chip pin on one of those (don't know which one).
1020000395
D/C: 2022.7.7
D/C: 2022.7.20

AnAwesomePossum, 2022-09-04:
How are you guys finding the date?

frutabruta, 2022-09-04:
> How are you guys finding the date?

Sticker on the bottom of the carboard box.

billru05, 2022-09-04:
frutabruta/ I looked under the microscope after I read your post but I did not see anything broken

frutabruta, 2022-09-04:
> frutabruta/ I looked under the microscope after I read your post but I did not see anything broken

It was probably a soldering flaw only on my board and the problem with the "bad series" will be something else.

shawnchr, 2022-09-04:
I grabbed the logs off my wifi and I had one that ran for 29 mins dauth its self from the router and now cannot reconnect although watching hooked up to a monitor I can see try and fail.
it runs sudo create.ap get command not found and then says it cannot find My network that 27 other devices are using.

The other will not connect at all but they both show network running 
nmcli dev wifi list

billru05, 2022-09-04:
frutabruta/ one of  your dates on yours is the same as mine 2022.07.20 and I can not get it to work, I can get the farthest with the Hurakan_image, but it breaks as soon as you start running commands from putty

billru05, 2022-09-04:
shawnchr/ are you using the CB1 board wifi or a USB dongle

shawnchr, 2022-09-04:
I was using the onbaord Wifi

bigtreetech, 2022-09-05:
Hello. We are very sorry, If you guys CB1 is STILL having WIFI issues even though V2.1.2, plz don't hesitate to email support01@bigtree-tech.com, we will try our best to help you guys out. Please note CB1 when sending emails , we will solve the problem for you in time.

looxonline, 2022-09-05:
> So it appears something before 2022.7.30 changed or bad components

Are you on facebook? If so get in touch with me on the BTT group. I'm an admin and I'll help you to get in touch with the right people and get sorted ASAP.

billru05, 2022-09-05:
So finally after emailing them on Aug 30th they answered my email but with no solution, here is a snip of the reply (For WIFI connection, an updated image file is currently being compiled, so it is recommended to use wired network connection first. Thanks for understanding) so base on this there is another image coming but I am not holding my breath.

macgirl, 2022-09-05:
> Can anyone confirm the dates on the CB1 that are working vs the ones that are not, this is my dates and nothing works: CB1 (1020000395 BIGTREETECH CB1_V2.1 1G D/C 2022.7.27 PI4B 1030000103 [...]

Mine have the same numbers as yours. None work.

looxonline, 2022-09-05:
> So finally after emailing them on Aug 30th they answered my email but with no solution, here is a snip of the reply (For WIFI connection, an updated image file is currently being compiled, so it is [...]

That's an old response in reference to the current latest image. Contact me on FB and I will get you sorted.

macgirl, 2022-09-05:
> That's an old response in reference to the current latest image. Contact me on FB and I will get you sorted.

Can't you sort this out right here? All of us would like or CB1s working.

looxonline, 2022-09-05:
> > > So finally after emailing them on Aug 30th they answered my email but with no solution, here is a snip of the reply (For WIFI connection, an updated image file is currently being compiled, so it is recommended to use wired network connection first. Thanks for understanding) so base on this there is another image coming but I am not holding my breath.
> > 
> > 
> > That's an old response in reference to the current latest image. Contact me on FB and I will get you sorted.
> 
> Can't you sort this out right here? All of us would like our CB1s working.

I don't work for them. I just help out with being an admin on their group. Sometimes they struggle a bit with the English barrier and then I lend a hand in my free time to bridge the gap. FB messenger is convenient for me because I throw the right people into the conversation and get it sorted ASAP.

looxonline, 2022-09-05:
> Can't you sort this out right here? All of us would like our CB1s working.

I am sure that the normal channels will also work. I am simply offering a shortcut for those who want to take it.

macgirl, 2022-09-05:
> 

Perhaps others will be kind enough to post the solution. I do not choose to use FB or their products.

looxonline, 2022-09-05:
> > 
> 
> Perhaps others will be kind enough to post the solution. I do not choose to use FB or their products.

The solution is that I am going to help you get your board replaced if it is not working. I am also on their discord server as an admin.

[61 more replies]