bigtreetech/CB1 #30: Does not boot at all PI4B/CB1

zentaur1ger, 2022-09-07:
Hello everyone,

I actually can't get it to work at all.
I have two PI4B and CB1 here.
I downloaded CB1_Debian11_bullseye_minimal_kernel5.16_20220902.img.xz, flashed it to an SD card and tried it in both devices.
On the CB1, LED2 starts to glow steadily as soon as you connect the power supply. After about 30 seconds, the ACT LED on the PI4B will start flashing regularly. So not in the way you know it from the PI when the SD card is accessed, but just regularly on off, on off.....
The POWER LED does not come on at all.

Of course I have also supplied the system.cfg with my WLAN credentials, but neither of the devices appear in the WLAN either.

Now I have repeated all this several times, also with different SD cards.
Now I think it's unlikely that two brand new devices are defective at the same time and rather assume I've overlooked something or skipped a step.

Can someone who boots from this combination write to me briefly how he got there?

Kind regards

Oliver

bigtreetech, 2022-09-08:
Hello.
1. "LED2 on CB1 starts to glow steadily as soon as you connect the power supply" is normal
2. If PI4B Adapter is used with CB1.  the PWR LED on PI4B Adapter never work (This is the case when used with CB1).  the ACT LED on PI4B Adapter will flashing every second in the current CB1 OS.
3. 
According to your description(ACT LED on the PI4B will start flashing regularly) CB1 should have been started now.

Can you connect a HDMI screen to check whether the screen can display the command line interface, then connect the keyboard to log in ( login as: biqu password: biqu), 
typing 
`sudo nmcli device wifi list`
to check all the WiFi that can be scanned,
and typing 
`sudo nmcli c`
to check the connected WiFi status. 

Then typing 
`cat /etc/scripts/wifi.log`
 to take pictures of the log information displayed. I will diagnose it

zentaur1ger, 2022-09-08:
[image]

Hi, thank you for your answer. You're right, that thing is really booting. May i suggest to change the behaviour of that ACT LED from constantly blinking to something that indicates access to the SD Card? Would at least make it easier to dignose problems.
I've wiped my WLAN password from that screenshot, but it is correct. 
Not sure what the CB1 does at that point, but in the logs of my router i do not find any login attempts from this device.

Kind Regards

Oliver Kalitowski

bigtreetech, 2022-09-09:
Hello
the result of `sudo nmcli device wifi list` shows that there are two WiFi of `TIDIRIUM-GATE`. Is your router mixed with 5G and 2.4G and uses the same WiFi SSID? Can you try to set the SSID of 2.4G and 5G separately. The WiFi on CB1 only supports the connection of 2.4G WiFi.

or you can typing
`sudo nmtui` -> `Activate a connection`  selet your WIFI SSID and typing password to manually connect WiFi to see if it can be connected normally.

zentaur1ger, 2022-09-09:
1. Set the SSID for the 2.4 G Network to TIDIRIUM-GATE2.
2. Edited system.cfg . The Device still tries to connect to TIDIRIUM-GATE instead of TIDIRIUM-GATE2
3. Changed the configuration with sudo nmtui, also does not work.
4. Wiped the SD Card and flashed the image again. Edited system.cfg to the correct SSID and password. Now it states "rc.local [1719] Failed to stop create_ap.service: Unit create_ap.service not loaded. sudo: create_ap: command not found
5. Result: Still not connecting to the wifi

EchoHeim, 2022-09-09:
Hello!
@zentaur1ger 

After you set the 2.4G wifi ssid to TIDIRIUM-GATE2, re-burn the system image to the sd card, edit system.cfg correctly, and send me the /etc/scripts/wifi.log for a look.

You can also just type 
`sudo nmcli dev wifi connect TIDIRIUM-GATE2 password YOUR_PASSWD ifname wlan0`
Then send the feedback as well.

Also, the error message `"rc.local [1719] Failed to stop create_ap.service: Unit create_ap.service not loaded. sudo: create_ap"` can be ignored, it is about turning on the wifi hotspot. This feature is not supported yet.

zentaur1ger, 2022-09-09:
I've been gone the other way around. I've renamed my 5G Network, so that all other devices in my home that prefers 2.4G stay online.
In addition i've changed the WLAN password to a simple value without special characters to avoid trouble with language settings.

But as requested, i've created a fresh SD Card, edited system.cfg accordingly but still get no good result.
Interestingly, when i use a wrong password intentionally, i can see failed login attempts on my router.
Also tried to restrict the WLAN security to WPA2 only, but that has also no effect.

[image]

[image]

EchoHeim, 2022-09-09:
What about connecting directly by entering the following command.
 `sudo nmcli dev wifi connect TIDIRIUM-GATE password 102030405060708090 ifname wlan0`

zentaur1ger, 2022-09-09:
[image]

GadgetAngel, 2022-09-10:
Hi, 

I have a Pi4B/CB1 also, and my LED2 light on the CB1 does not light up after I connect the USB-C cable to my computer.  Also the ACT light never starts to blink.

I have downloaded the latest version of firmware to a 32GB micro-SD card (https://github.com/bigtreetech/CB1/releases) and added my WIFI info to the system.cfg file.  

But I think because the LED2 light on the CB1 never lights up that the CB1 board is bad.  I also bought another CB1 and connected the second CB1 to the Pi4B adapter and the second CB1's LED2 light will light up so this tells me the first CB1 is a bad unit.  How do I get it replaced?

EchoHeim, 2022-09-12:
Hi
@zentaur1ger 
The prompt message indicates that the wifi service is not fully loaded, you can do the following.

Add the following two lines to the `/etc/NetworkManager/NetworkManager.conf` file
```
[device]
wifi.scan-rand-mac-address=no
```

Then type in the terminal
`sudo systemctl restart NetworkManager`
for restarting the network service.

then you can connect to the ssid by
`sudo nmcli dev wifi connect TIDIRIUM-GATE password 102030405060708090 ifname wlan0`

It is recommended that the SSIDs of 2.4G and 5G wifi are named differently to differentiate between them.

EchoHeim, 2022-09-12:
hello，
@GadgetAngel 
If the board is confirmed to have problems, directly contact our after-sales service to arrange for replacement.

zentaur1ger, 2022-09-14:
@EchoHeim 
Still got the exact same message.
The wifi with SSID TIDIRIUM-GATE is only 2.4G, my 5G network is named different at the moment.

zentaur1ger, 2022-09-14:
@GadgetAngel I'm using monitor and keyboard all the time.
And now i've finally managed to bring that damn thing online.
What i've done is

1. changed wifi security on my router again down to WPA2 (did that before without effect)
2. created a root account on the CB1
3. logged in as root
4. sudo nmcli dev wifi connect TIDIRIUM-GATE password 'mypassword' ifname wlan0

that finally worked.
Now i also changed the keyboardlayout to german, changed sources.list to mirrors that suit me better and made an apt upgrade.
Changed back wifi security on my router to WPA3, the CB1 is still able to log in in my network automatically after reboot and also that "rc.local [1719] Failed to stop create_ap.service: Unit create_ap.service not loaded. sudo: create_ap" message is gone.

Within the upgrade process i've noticed an error message```
 dpkg: warning: parsing file '/var/lib/dpkg/status' near line 8161 package 'linux-u-boot-h616-current':
 missing 'Description' field
```
and also 
`ln: failed to create hard link '/boot/initrd.img-5.16.17-sun50iw9.dpkg-bak' => '/boot/initrd.img-5.16.17-sun50iw9': Operation not permitted`

but, it is finally working, which is nice. BUT for a non linux native this is quite a confusing process. So there is at least "some" room for improvements.

closed 2022-09-14 by zentaur1ger

davbcr, 2022-09-15:
> > @EchoHeim Still got the exact same message. The wifi with SSID TIDIRIUM-GATE is only 2.4G, my 5G network is named different at the moment.
> 
> Have you thought of using a Monitor and keyboard and logging into the CB1 and check the "Security" option of the WIFI via the `nmtui` tool. I found that to get my CB1 connected to my WIFI I had to change the "Security" option from "WPA3 Personal" to "WPA and WPA2 Personal" before my CB1 would connect to my WIFI

In my case I tryed that too, but the monitor doesn't show anything and i cannot proceed.
I'm stuck 'cause Wifi doesn't connect to my WPA2 router.
Is there a way to force switching to WPA2 security modifying any of the SD card files directly?

zentaur1ger, 2022-09-15:
Does the ACT light blink on your board?

davbcr, 2022-09-15:
> […](#)

They are.

GadgetAngel, 2022-09-15:
> In my case I tryed that too, but the monitor doesn't show anything and i cannot proceed. I'm stuck 'cause Wifi doesn't connect to my WPA2 router. Is there a way to force switching to WPA2 security [...]

Are you using this image : https://github.com/bigtreetech/CB1/releases labelled "CB1_Debian11_bullseye_minimal_kernel5.16_20220902.img.xz"?  With this image your monitor should work

docman11, 2023-05-08:
I have same problem with BTT PI4B with CB1. After flashing microsd with image - I can boot only 1 time (ACT led start blinking after about 2-3 min when coonected to WiFi, and works good (web interface & SSH). But after shut down or reboot - stay off and eats 110mA....Tried other microSD's / other power supply / different OS images 2.3.2 / 2.3.1 / 2.2.1 / minimal / full OS images,  diverse image flashers balena eatcher, raspberry img writer, rufus... - same problem... Only formatting microsd and flash image again help to start it again for 1 time..... Now I don't even understand - it's hardware or software problem in CB1..... not solved...