bigtreetech/CB1 #31: Waveshare CANHAT compatibility

shogowarrior, 2022-09-08:
How do I install CANHAT from waveshare on `CB1`. I believe it needs 
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=1000000
```

but not sure how to get the overlays working for this build.

PeterHouseJr, 2022-09-16:
Also trying to accomplish the same thing.

Where are device overlays installed on the CB1?

tnap1979, 2022-09-20:
Spi doesn't work on cb1 from what I believe..
And neither does wifi... lmao
Such a wasted product line

frutabruta, 2022-09-21:
> Spi doesn't work on cb1 from what I believe.. And neither does wifi... lmao Such a wasted product line

SPI does work, but it can only be used for ADXL345 so far. Wi-fi does work too, with the 2.1.2 firmware and nondefective board.

tnap1979, 2022-09-21:
Yea, I finally got a fixed version cb1 with the new ldo regulator yesterday, got it setup lastnight on the latest slim image... so gar so good..

shogowarrior, 2022-09-21:
That's great to know! WiFi has been working without any issues for me but not sure how to setup overlays. I wish they had a generic armbian build than the custom one

emersonsc, 2022-09-27:
> > Spi doesn't work on cb1 from what I believe.. And neither does wifi... lmao Such a wasted product line
> 
> SPI does work, but it can only be used for ADXL345 so far. Wi-fi does work too, with the 2.1.2 firmware and nondefective board.

Ok... How? And using BTT image?

frutabruta, 2022-09-27:
> > > Spi doesn't work on cb1 from what I believe.. And neither does wifi... lmao Such a wasted product line
> > 
> > 
> > SPI does work, but it can only be used for ADXL345 so far. Wi-fi does work too, with the 2.1.2 firmware and nondefective board.
> 
> Ok... How? And using BTT image?

https://github.com/bigtreetech/CB1/discussions/9#discussioncomment-3541418