bigtreetech/CB1 #32: Eth0 is not working

blandown, 2022-09-10:
Hi, I installed the latest minimal version and tried to access using ethernet cable, but didn't work.

[image]

When plugging in the ethernet cable, the leds do not light up.

BeigeAnimal, 2022-09-11:
Known issue read the release page

billru05, 2022-09-11:
Here is the Snip---V2.1.2 The ethernet driver of this version is still under debugging, You can use the kernel4.9 image first if you prefer to use the Ethernet rather than WiFi---

looxonline, 2022-09-27:
Fixed with a new release coming in about two days.

creatron, 2022-10-01:
The ETH fix released works, but it send zillions of messages to /var/log/kern.log. Please remove the logs xxx phydev->speed=100  full

AChorostian, 2023-02-22:
It's still not fixed in 2.3.1

looxonline, 2023-02-22:
> It's still not fixed in 2.3.1

It was fixed but there is a new version of the SOC that needs a slight tweak to the driver. A new image will be released soon with the updated driver.

AChorostian, 2023-02-22:
Is there available any OS that supports ethernet port on Manta M8P v1.1 ?
I just tried Hurakan_Debian11_Mainsail_kernel4.9_20220912
Ethernet again doesn't work.

looxonline, 2023-02-22:
> Is there available any OS that supports ethernet port on Manta M8P v1.1 ? I just tried Hurakan_Debian11_Mainsail_kernel4.9_20220912 Ethernet again doesn't work.

As I mentioned, the updated OS will be coming in a few days.