bigtreetech/CB1 #37: Can't Flash MP4 - Invalid DFU suffix signature

acosano, 2022-09-27:
I spend now the whole day trying to flash (update) klipper too the lates version but with no sucess and  Ireally don't know anymore the to take board out and put the old one. I keep getting this and  Idon't know what to do.

###### Stop klipper.service ...
[✓ OK] Stop klipper.service successfull!
  Flashing out/klipper.bin to /dev/serial/by-id/usb-Klipper_stm32g0b1xx_hurakan-if00
Entering bootloader on /dev/serial/by-id/usb-Klipper_stm32g0b1xx_hurakan-if00
Device reconnect on /sys/devices/platform/soc/5200000.usb/usb2/2-1/2-1.1/2-1.1:1.0
sudo dfu-util -p 2-1.1 -R -a 0 -s 0x8002000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/serial/by-id/usb-Klipper_stm32g0b1xx_hurakan-if00: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper_stm32g0b1xx_hurakan-if00

make: *** [src/stm32/Makefile:90: flash] Error 255
>>>>>> Flashing failed!
>>>>>> Please read the console output above!

frutabruta, 2022-09-27:
You can try flashing from microSD: get the file /home/your_username/Klipper/out/klipper.bin and copy it to microSD card, rename the fire to firmware.bin and put the card into M4P to the slot marked as MCU microSD. Then power cycle the board and it should be flashed.

acosano, 2022-09-27:
Hi Thanks

> You can try flashing from microSD: get the file /home/your_username/Klipper/out/klipper.bin and copy it to microSD card, rename the fire to firmware.bin and put the card into M4P to the slot marked as MCU microSD. Then power cycle the board and it should be flashed.

I been using the  Microsd card slot on the board but same no sucess. I'm not aware that there is a separated slot for the MCU can you please where it is located as  Ican't see it in the manual.

frutabruta, 2022-09-27:
> Hi Thanks
>
>
> I been using the Microsd card slot on the board but same no sucess. I'm not aware that there is a separated slot for the MCU can you please where it is located as Ican't see it in the [...]

It seems to be the normal size SD card slot next to microSD slot.

acosano, 2022-09-27:
That work. It is just bad documented. Ifyou look on their page is market in blue saying that the MCU can be update and micro SDcard is market in bluenot the normal size SD. 
Anyway thanks for your help you save me changing boards.

closed 2022-09-27 by acosano