bigtreetech/CB1 #40: V2.2.0 ssh disabled?

davbcr, 2022-09-29:
After fresh flashing a SD card with CB1_Debian11_bullseye_minimal_kernel5.16_20220929.img just released i put into CB1 and finally i get it connected to my WPA2 network, but SSH connection (port 22) by Putty is refused.
it's still visible into "connected device" tab of my router but i cannot have access to it.
Is SSH disabled?
How can i get the SSH again without being able to see anything on screen? Screens i have do not support the resolution

davbcr, 2022-09-29:
Port 22?
If that's so the i really nave no clue why isn't working and refuse connection.

thisiskeithb, 2022-09-29:
> Port 22?

Yup.

davbcr, 2022-09-30:
Stuck again with WiFi not connecting issue.
I erased SD card again and flashed the new release V2.2.0 on 30.09, and it starts not connecting to WiFi again.

frutabruta, 2022-09-30:
I had similar problem once with different board. The power supply was so bad it corrupted SD card and turned it into ’filesystem is read only' state which effectively disabled SSH. Try different SD card. Are you sure you are trying the correct IP address? 
I can connect via SSH on V2.2.0 without any problem.

davbcr, 2022-09-30:
it cannot be my case, i'm using a 75W external power supply that give upt to 10A since the first try and SD card was fresh in the beginning.
By the way i'll try with a different SD.

NAPCAL, 2022-09-30:
Here is how I fixed my CB1 issue connecting to my eero 6 Pro mesh WiFI. 
- I found the file system.cfg on the SD card after putting the 2.2.0 minimal image on it. 
- Edited the 1st WiFi entries to match my SSID and password and saved the file.
- Put the SD card into my Manta M8P and powered it up. Waited for a good five min before proceeding.
- Checking my eero app, no new connections showed up.
- I then plugged in an ethernet cable to get access via ssh to the CB1. Note: BTT image defaults are User Name: biqu | Password: biqu
- Searched the web for 'Debian 11 setup WiFI via command line', and found this site. https://linuxhint.com/3-ways-to-connect-to-wifi-from-the-command-line-on-debian/ 
- After trying the 1st one listed, with no help.
- The second one got me the info I needed, and the default for the settings in the system.cfg at the start ended up being set for WPA3 Personal; changing this to WPA & WPA2 Personal fixed the connection issue.

- Looking at my eero app to get the IP address, I noticed that the WiFi strength was low (not an issue with my CM4's), and I am in the same room with one of my eero 6 Pro, but I did have a CM4 WiFi external Antenna that I put on the CB1 and is now showing full strength WiFi signal.

davbcr, 2022-10-06:
something must be broken on my boards
no SD is being read by it and no images appear on any monitor i tried.
I contacted the service and i really hope the will propose to change it for me.

closed 2022-10-07 by davbcr