bigtreetech/CB1 #43: Not showing up on the network after fresh install

jamesmac10, 2022-10-01:
I downloaded the latest klipper image from your repository and flashed it using pi imager. put the sd card in the rpi 4 adapter with cb1 and plugged it in and there is a green solid led on the cb1 and a flashing green led on the rpi 4 adapter but nothing new is showing in the network. Oh and I also edited system.cfg and added my wireless SSID name (case sensitive) and password. Any thoughts?

davbcr, 2022-10-05:
I state that i never get my CB1 up and running so far with any FW release but reading here and there i can tell you that if the green led is flashing then the CB1 works but it's unable to connect the WiFi probably because it's default set is a WPA3 security.
You need a monitor (and here is where i'm stuck at) to plug the CB1 into and go via [nmtui ](https://linuxhint.com/3-ways-to-connect-to-wifi-from-the-command-line-on-debian/) (3rd method) tool to change to WPA2 probably.

Danielos94, 2022-10-05:
I have same issue, after changing router setting to WPA3 still no connection..

ebraiman, 2022-10-06:
@jamesmac10 This might help ... https://youtu.be/dcsyyR_6bYE