bigtreetech/CB1 #44: GPIO Access Denied

S1L3NTANG3L, 2022-10-02:
Moonraker is currently not able to access the gpio pins.
[gpio.py:_request_gpio()] - Unable to init gpiochip0:gpio6.  Make sure the gpio is not in use by another program or exported by sysfs.
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 84, in _request_gpio
    chip = self._get_gpio_chip(pin_params['chip_id'])
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 40, in _get_gpio_chip
    chip = self.gpiod.Chip(chip_name, self.gpiod.Chip.OPEN_BY_NAME)
PermissionError: [Errno 13] Permission denied
2022-10-02 13:02:25,273 [moonraker.py:add_warning()] - Failed to load power device [power light_strip_2 ]
Error parsing option (pin) from section [power light_strip_2 ]

Michelinu, 2022-10-02:
CB1 uses a custom IO arrangement, check my post here: https://github.com/bigtreetech/CB1/issues/14#issuecomment-1264658542

Just take a `PC` PIN and use my "guide"; I tried with `PC8`, `PC10 `and `PC15` and they works

S1L3NTANG3L, 2022-10-02:
Hi thanks for the feedback I tried both the PC pins directly and the SOPine Gpio Alternate. Still getting a permission denied error. What version of the CB1 OS are you running?

2022-10-02 19:28:36,195 [gpio.py:_request_gpio()] - Unable to init gpiochip0:gpio74.  Make sure the gpio is not in use by another program or exported by sysfs.
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 84, in _request_gpio
    chip = self._get_gpio_chip(pin_params['chip_id'])
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 40, in _get_gpio_chip
    chip = self.gpiod.Chip(chip_name, self.gpiod.Chip.OPEN_BY_NAME)
PermissionError: [Errno 13] Permission denied
2022-10-02 19:28:36,195 [moonraker.py:add_warning()] - Failed to load power device [power light_strip_1]
Error parsing option (pin) from section [power light_strip_1]
2022-10-02 19:28:36,197 [gpio.py:_request_gpio()] - Unable to init gpiochip0:gpio79.  Make sure the gpio is not in use by another program or exported by sysfs.
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 84, in _request_gpio
    chip = self._get_gpio_chip(pin_params['chip_id'])
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 40, in _get_gpio_chip
    chip = self.gpiod.Chip(chip_name, self.gpiod.Chip.OPEN_BY_NAME)
PermissionError: [Errno 13] Permission denied

Michelinu, 2022-10-02:
I'm using latest Klipper image of CB1 OS, the V2.2.0. I just tried and I javethe same problem.
If I set the type as `gpio`, I get the permission denied error; if I configure it as `output_pin` in Klipper and then set it up as `klipper_device` in Moonraker, it works.

I think that it's just a permission problem. Tried with the solution in this post but it didn't work: https://github.com/Arksine/moonraker/issues/249#issuecomment-945578614

S1L3NTANG3L, 2022-10-02:
Okay so you used the PC pins in klipper?
aka
[output_pin led1_pin]
pin: PC14
pwm: False
value: 0
and
[power light_strip_1] #output_7
type: klipper_device
object_name: output_pin led1_pin

Damn this is a good work around but it still has the downside of not being able to turn the printer on and off :'(

S1L3NTANG3L, 2022-10-02:
Hmmm tried [](https://github.com/Arksine/moonraker/issues/249#issuecomment-945578614) and now the error looks different:
2022-10-02 22:38:09,928 [gpio.py:_request_gpio()] - Unable to init gpiochip0:gpio24.  Make sure the gpio is not in use by another program or exported by sysfs.
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/components/gpio.py", line 97, in _request_gpio
    line.request(**args)
OSError: [Errno 517] Unknown error 517

Michelinu, 2022-10-02:
> 

Assuming that you used the Klipper image, so you had the linux MCU host already install or you did it by yourself, you must have:

- In your `printer.cfg`:
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[output_pin led1_pin]
pin: rpi:gpio78
```

- In your `moonraker.conf`
```
[power light_strip_1]
type: klipper_device
object_name: output_pin led1_pin
```

Just checked and it corresponds: GPIO78 -> GPIO17_PC14 -> PIN n.11
Having a power device in Moonraker gets pretty useless in this case, as you said you won't be able to control it if Klipper is off.

You can try to set it only in your `printer.cfg`; if it works as expected, you can use it in this way until there will be a fix for Moonraker

Michelinu, 2022-10-02:
> Hmmm tried [ Power GPIO permission denied #249 ](https://github.com/Arksine/moonraker/issues/249#issuecomment-945578614) and now the error looks different: 2022-10-02 22:38:09,928 [gpio.py:_request_gpio()] - Unable to init gpiochip0:gpio24. Make sure the gpio is not in use by another program or exported by sysfs. Traceback (most recent call last): File "/home/biqu/moonraker/moonraker/components/gpio.py", line 97, in _request_gpio line.request(**args) OSError: [Errno 517] Unknown error 517

You can't use `GPIO24` in our CB1

Michelinu, 2022-10-02:
I just found the solution.

Add the following lines in your `moonraker.conf`:
```
[power my_device]
type: gpio
pin: gpio78
```

Create a new udev rule as `/etc/udev/rules.d/60-gpiod.rules` :
```
# udev rules for gpio port access through libgpiod
SUBSYSTEM=="gpio", KERNEL=="gpiochip[0-4]", GROUP="biqu", MODE="0660"
```

The solution I posted was using the "gpio" group, that does not exist in our CB1 image.
So by just using "biqu" as group, it works!

S1L3NTANG3L, 2022-10-02:
Okay so [this](https://github.com/Arksine/moonraker/issues/249#issuecomment-945578614) and [this](https://wiki.pine64.org/wiki/SOPine_GPIO_Alternate_Functions_Table) in combination with [this](https://github.com/mkuf/prind/issues/8#issue-1124760831) seem to be allowing me to add 
[power light_strip_2 ] 
type: gpio
pin: gpio74
off_when_shutdown: True
initial_state: off
to my moonraker.conf and not get any error after a system reboot.
I'll quickly just go and check pin voltages and make sure its working.

Michelinu, 2022-10-02:
> Okay so [this](https://github.com/Arksine/moonraker/issues/249#issuecomment-945578614) and [this](https://wiki.pine64.org/wiki/SOPine_GPIO_Alternate_Functions_Table) in combination with [...]

If you don't get any error, it should work.
I use the CB1 with the Pi4 adapter and I have 1.8V in the PIN when it's "powered".

Remember that `GPIO74` -> `GPIO22_PC10` -> `PIN n.15` in the 40-pin connector

S1L3NTANG3L, 2022-10-02:
Lovely I'm also getting around 1.8V output on high.
So the working pins on the M8P are:
`GPIO17`->`PC14`->`GPIO78`
`GPIO27`->`PC12`->`GPIO76`
`GPIO22`->`PC10`->`GPIO74`
`GPIO6`->`PG6`->`GPIO198`
`GPIO23`->`PC11`->`GPIO75`
`GPIO24`->`PC9`->`GPIO73`
`GPIO19`->`PC6`->`GPIO70`
`GPIO26`->`PC15`->`GPIO79`
This is the ones that I quickly tested.

Michelinu, 2022-10-02:
> Lovely I'm also getting around 1.8V output on high.
> So the working pins on the M8P are:
> `GPIO17`->`PC14`->`GPIO78`
> `GPIO27`->`PC12`->`GPIO76`
> `GPIO22`->`PC10`->`GPIO74`
> `GPIO6`->`PG6`->`GPIO198` [...]

You already tried more pins that me!
I did the "research" because maybe in the future I would need to use some of them.

The "disappointment" is that people have been asking Biqu/BTT for the GPIO pinout since weeks/months.
There's still nothing of ufficial, except for that scheme I posted that you can find only in M4P user manual but doesn't tell you what `GPIO<number>` to use!

S1L3NTANG3L, 2022-10-02:
Yeah my little list atleast references the current m8p pinout, weird that 75 aka gpio23 gives a error 16 device busy

S1L3NTANG3L, 2022-10-02:
Now I just hope I can use this list on the radxa zero, having the same pinout issue there.

S1L3NTANG3L, 2022-10-02:
That's supposed to be gpio8 according to there m8p pinout. This manual is useless. Well I have 8 functioning pins which is more than enough for my purposes. Thanks again for the help

yzeroy, 2022-10-11:
Hi all,

`pi@Ender:~$ uname -a`
`Linux Ender 4.9.255-sun50iw9 #1 SMP PREEMPT Tue Mar 1 23:28:34 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

I wanted to use GPIOs for controling 3D printer relay. And it is finaly fixed by these steps.

`Sudo nano /etc/udev/rules.d/60-gpiod.rules`
and add 
`# udev rules for gpio port access through libgpiod`
`SUBSYSTEM=="gpio", KERNEL=="gpiochip[0-4]", GROUP="USERNAME ", MODE="0660"`

Thanks @Michelinu 

next 
`sudo nano /etc/systemd/system/moonraker.service`

and add
`[Service]`
`Type=simple`
`User=username`
`Group=username`

Reboot