bigtreetech/CB1 #45: Can't ssh with default password, permission denied

Shahid-Manzar, 2022-10-05:
Hi, I am new to 3D printing but have experience with programming and engineering tools, relatively novice to GitHub as well so forgive my ignorance.

I am using M8P with CB1. Firmware is klipper 2.2.0

Reinstalled onto SD card 3 times so far with no luck, tried with ethernet and wifi. The default password "biqu" (without quotations) does not work, also tried other passwords including password, raspberry, whatever else came to mind, etc.

Does anyone know what's going on? I put a screenshot of the terminal response below

[image]

frutabruta, 2022-10-05:
Try:
ssh biqu@192.168.1.148

Shahid-Manzar, 2022-10-05:
Oh wow, that fixed the issue. I feel dumb, thanks!

closed 2022-10-05 by Shahid-Manzar