bigtreetech/CB1 #46: V2.2.0 - CB1_Debian11_bullseye_Klipper_kernel5.16_20220929.img.xz (32GB Card Needed)

o1hitman1o, 2022-10-05:
It should be noted that you need to use a 32GB SD card if you are going to be flashing the V2.2.0 CB1_Debian11_bullseye_Klipper_kernel5.16_20220929.img.xz image that was uploaded.
I was using a 16GB card and it keep failing.

thisiskeithb, 2022-10-07:
Something was probably wrong with your 16GB SD card since `CB1_Debian11_bullseye_Klipper_kernel5.16_20220929.img.xz` works fine on an 8GB SD card. There's even room to spare:

[image]

I'd recommend using the official [SD Memory Card Formatter tool](https://www.sdcard.org/downloads/formatter/) and reformatting the 16GB card.

o1hitman1o, 2022-10-07:
Hmmm... Not sure.  I did try multiple SD cards but it only completed flash on the 32 GB card.

looxonline, 2022-10-15:
16GB card works just fine.

o1hitman1o, 2022-12-30:
Must work then.

closed 2022-12-30 by o1hitman1o