bigtreetech/CB1 #50: CB1_Debian11_bullseye_Klipper_kernel5.16_20220929.img.xz unable to find

Wakinyantanka, 2022-10-16:
where is this image. I'm not finding anywhere to download.
thanks

billru05, 2022-10-17:
its in  V2.1.2 under assets

Wakinyantanka, 2022-10-17:
I see it there but it's greyed out. How do you go about downloading it? The minimal and hurakan kernels are in the asset section and can be downloaded but not the Klipper kernal.

Wakinyantanka, 2022-10-17:
Sorry if it's obvious, I'm new to this.

billru05, 2022-10-17:
I am just trying to help; I was not sure I was understanding what you wanted, just got to the OS page, from the V2.2.0 scroll down to Assets, if you do not see the files under assets then click assets and you will see the file you are looking for [CB1_Debian11_bullseye_minimal_kernel5.16_20220929.img.xz}, click the file and it will download

Wakinyantanka, 2022-10-17:
I appreciate the help. 
I did as you instructed and I'm still not seeing the latest Klipper kernal. The minimal kernal is there a long with the hurakan but that's it.

billru05, 2022-10-17:
This is the link, see if this works: https://github.com/bigtreetech/CB1/releases/download/V2.2.0/CB1_Debian11_bullseye_minimal_kernel5.16_20220929.img.xz

Wakinyantanka, 2022-10-17:
Unfortunately it shows the minimal kernel and the hurakan but not the Klipper kernel. 
Let me ask this, if I download the older Klipper kernel and update it through my terminal via kiauh will that then be the latest update?

looxonline, 2022-10-22:
> Unfortunately it shows the minimal kernel and the hurakan but not the Klipper kernel. Let me ask this, if I download the older Klipper kernel and update it through my terminal via kiauh will that then be the latest update?

There is no such thing as a klipper kernel. The hurakan has klipper pre-installed and the minimal does not. Personally I always use minimal and KIAUH to install to my liking.