bigtreetech/CB1 #53: 3.5 inch rpi lcd on CB1

DmitriyAD, 2022-10-24:
How do I start the screen 3.5 inch rpi lcd on the BIGTREETECH PI4B Adapter V1.0 and CB1 ?

looxonline, 2022-10-26:
The CB1 does not support DSI. You need to use an HDMI display. Future compute module boards will support DSI.

DmitriyAD, 2022-10-26:
This screen is not a DSI he's a RPI
http://www.lcdwiki.com/3.5inch_RPi_Display

DmitriyAD, 2022-10-26:
> The CB1 does not support DSI. You need to use an HDMI display. Future compute module boards will support DSI.

This screen is not a DSI he's a RPI
http://www.lcdwiki.com/3.5inch_RPi_Display

looxonline, 2022-10-26:
Ah, it's an SPI display. Some CB1 boards will support it while others will not. It depends on what version you have. Some used 1.8V on the SPI lines to provide GPIO at a range of voltages for different applications; however, this was later adjusted so that the SPI lines operate at 3V3 to provide compatibility with devices like this. I believe the V2.2 boards have the 3V3 lines.

looxonline, 2022-10-26:
> > Ah, it's an SPI display. Some CB1 boards will support it while others will not. It depends on what version you have. Some used 1.8V on the SPI lines to provide GPIO at a range of voltages for different applications; however, this was later adjusted so that the SPI lines operate at 3V3 to provide compatibility with devices like this. I believe the V2.2 boards have the 3V3 lines.
> 
> So I will not be able to start this screen)?

It all depends on what version CB1 you have. Also, I have not tested whether the debian distro has the display and touch controller drivers for that display. I would assume that it must have them or that they can be installed though.

looxonline, 2022-10-26:
> It all depends on what version CB1 you have. Also, I have not tested whether the debian distro has the display and touch controller drivers for that display. I would assume that it must have them or [...]

I actually have a screen like that so I may test it this weekend.

DmitriyAD, 2022-10-26:
> I actually have a screen like that so I may test it this weekend.

I would be very grateful

vdd9474a, 2023-03-09:
Is there some news about this issue? I have a V2.2 CB1 and I also would like to use this kind of TFT screen

Pneumanifest, 2023-03-10:
> 

There's a new TFT35-spi screen that is designed for use with the CB1 and it works great... looks like it's currently sold out but here's the link https://biqu.equipment/products/bigtreetech-tft35-spi-v2-1-touchscreen-io2can-module