bigtreetech/CB1 #54: Uart not Working

pettard, 2022-10-26:
Hi, Uart is not possible to get working with the latest image.
Is there anyway to get this working again?

Haxk20, 2022-11-15:
UART is working correctly. Recheck if its connected correctly and your SDcard is plugged in.

Avasty, 2023-07-23:
I have trouble setting it up as well...

Pneumanifest, 2023-07-23:
In the BoardENV.txt file change
"console=display" to "console=serial" 
and use
"/dev/ttyS0" instead of "/dev/ttyAMA0"

Avasty, 2023-07-23:
That's what i have...

Avasty, 2023-07-23:
Exactly as you written, connected VIA 5 pins on board between **BigTreeTech Pi V1.2** and **BTT Pico V1.0** and still cannot connect to board

Avasty, 2023-07-23:
I think that problem might be somewhere else... https://pastebin.com/0xFdvyst here's klippy.log if you want to give it a try

Pneumanifest, 2023-07-23:
It's showing a connection time out, so no response. Have you tried swapping Rx and tx on one side?

Avasty, 2023-07-24:
Yes i did