bigtreetech/CB1 #61: Request to add devicetree overlay for Manta M4P's SoC fan

AntonisKl, 2022-11-13:
Hi,

I plugged in a 24v fan on the SoC fan socket (internal pin PC15) of the Manta M4P. The fan is not working because apparently an overlay is needed that makes the H616 SoC communicate with the fan's GPIO to turn it on/off depending on the SoC's temperature.
An example of this kind of overlay for Raspberry Pi can be found here: https://raw.githubusercontent.com/raspberrypi/linux/rpi-4.19.y/arch/arm/boot/dts/overlays/gpio-fan-overlay.dts .
It would be nice to have something like this adjusted for CB1.

AntonisKl, 2022-11-13:
For anyone who wants the SoC fan to be functional in the meantime, I am using this Bash script to do it (modified from [this source](https://raspmer.blogspot.com/2019/02/control-cooling-fan-with-gpio-fan.html))
```console
#!/bin/bash

GPIO_PIN=79 # corresponding pin name: PC15

HIGH_TEMP=50000
LOW_TEMP=45000

function get_temp() {
  cat /sys/class/thermal/thermal_zone0/temp
}

function gpio_enable() {
  echo $1 > /sys/class/gpio/export
}

function gpio_disable() {
  echo $1 > /sys/class/gpio/unexport
}

function gpio_set_output() {
  echo out > /sys/class/gpio/gpio$1/direction
}

function gpio_set_input() {
  echo in > /sys/class/gpio/gpio$1/direction
}

function gpio_set_on() {
  echo 1  > /sys/class/gpio/gpio$1/value
}

function gpio_set_off() {
  echo 0 > /sys/class/gpio/gpio$1/value
}

function cleanup() {
  QUIT=1
}

QUIT=0

#echo "Enable GPIO ${GPIO_PIN}"
gpio_enable $GPIO_PIN

#echo "Set GPIO ${GPIO_PIN} Output"
gpio_set_output $GPIO_PIN

#echo "set GPIO ${GPIO_PIN} on"
#gpio_set_on $GPIO_PIN

LASTSTATUS=0

trap cleanup SIGINT SIGTERM

while [ 1 ]; do
  cputemp=$(get_temp)
  #echo "CPU TEMP = $cputemp"
  if [ $LASTSTATUS = 0 ]; then
    if [ $cputemp -gt $HIGH_TEMP ]; then
      #echo "FAN ON"
      #echo "CPU TEMP = $cputemp"
      gpio_set_on $GPIO_PIN
      LASTSTATUS=1
    fi
  else
    if [ $cputemp -lt $LOW_TEMP ]; then
      #echo "FAN OFF"
      #echo "CPU TEMP = $cputemp"
      gpio_set_off $GPIO_PIN
      LASTSTATUS=0
    fi
  fi
  if [ "$QUIT" = "1" ]; then break; fi
  sleep 1
done

#echo "set GPIO ${GPIO_PIN} off"
gpio_set_off $GPIO_PIN

#echo "Disable GPIO ${GPIO_PIN}"
gpio_disable $GPIO_PIN
```
The process to make this script run at startup is this:
1. Create a file named `soc_fan.sh` in `/home/biqu/`
2. Edit it by running `nano soc_fan.sh` and copy-paste the above script
3. Change its permissions by running `sudo chmod 755 soc_fan.sh`
4. Edit `/etc/rc.local` by running `sudo nano /etc/rc.local` and add the line `/home/biqu/soc_fan.sh &` before `exit 0`

looxonline, 2022-11-22:
No device overlay needed. It can be controlled by klipper simply by exporting the pin (as you do in your script) and then adding the udev 60 rule. There is another issue in the list here which details the udev rule. Of course you need to add the CB1 as an MCU within klipper too.