bigtreetech/CB1 #63: Reduced wireless performance on new image

zhang-benj, 2022-11-21:
The latest V2.2.1 image released on 11-20-22 has far worse wireless performance, in my case, it will not pick up the SSID of a router located 15 feet away, nor will it connect if a connection profile is set up manually. Switching back to the older V2.2.0 image from 9-30-22 fixes the ability to connect to the same network that could not be connected to in the new image.

zhang-benj, 2022-11-21:
Some extra clarification: this issue occurs both with using nmtui and the wifi configuration script. Leaving the wifi connection enabled will also cause duplicate connection profiles to accumulate, it seems like a new one is added every time the system reboots.

Here's the wireless configuration
[image]

aodxxl, 2022-11-27:
I can confirm that the wifi performance is terrible. Isuing commands from PC, arrive with delays and the video stream from the webcam is unusable. Sometimes i get stable 15FPS and sometimes 0 FPS. The connection is stable but the bandwidth drops and picks up again. I tried lowering the resolution, changing bitrate and nothing helps. The printer is 4 m away from the router with a thin brick wall between. 

Not sure exactly what is causing this. Also using the latest Image.

Using a raspberry PI 3B+ i dont have this issue. Everything happends instantly. And keep in mind that the PI 3B+ uses an internal wifi antenna as oposed to the CB1 external antenna.

emersonsc, 2023-01-24:
Interesting. My brand new M4P + CB1 using 2.2.1 with zero issue. Router is on the other side of the house so I'm going through multiple walls. I've been transferring large gcode files onto it and doing updates with zero issue.