bigtreetech/CB1 #64: Enabling USB via config.txt

atcproxys, 2022-11-21:
Attempting to use a touch screen with the cb1. They say to change the config.txt file which is never created on the current images. Adding it manually doesn't enable USB either. 

support said:
Hello, this file is in your system settings. A USB 2.0 Hub is loaded on the Pad5 board. In order to save power, the USB port of CM4 is disabled by default. If you need to enable it, you need to add the following contents to the config.txt file:

dtoverlay=dwc2,dr_ mode=host

Does anyone know how to enable USB when the config file doesn't exist?

emersonsc, 2022-11-22:
You have to edit the config file when the SD card is plugged into a computer, not when its in the CB1. The config file is located on the small FAT32 partition.

atcproxys, 2022-11-22:
> FAT32 partition

Hi,

That is where I'm looking for the config.txt but it doesn't exist.

Drive contents:
[https://ibb.co/W5gdFMW](url)

emersonsc, 2022-11-23:
i believe its the system.cfg the same place where you add lines for the wifi and other variables.

atcproxys, 2022-11-28:
> The USB on *CB1* is enabled by default. If your touchscreen does not
> work, the problem is somewhere else. The config.txt settings only apply for
> *CM4*.

So this module is not compatible with a raspberry pi touchscreen then

emersonsc, 2022-11-28:
> So this module is not compatible with a raspberry pi touchscreen then

Which one? What specific make/model screen are you trying to use?

frutabruta, 2022-11-28:
No, displays connected via DSI interface are not supported on CB1 (BTT Pad 5 and official 7" Raspberry display). You can use only HDMI connected displays (some SPI displays might be added later). 
From BTT Pad 5 product webpage:
>Please note: 
>2, The Raspberry Pad 5 couldn't work with CB1, please be aware.
>https://biqu.equipment/products/bigtreetech-raspberry-pad-5

BTT HDMI 5 works fine.

atcproxys, 2022-11-28:
This screen uses hdmi for signal and usb for power and touch screen input.
DSI is not used for it which means there should be a touch screen driver
fix?

frutabruta, 2022-11-28:
@atcproxys so it is a HDMI display. So you should be able to see the image. If the touchscreen does not work, you might try this `sudo apt install xserver-xorg-input-libinput`
If that did not help, the toushcreen uses some non sstandard driver.

atcproxys, 2022-11-28:
Looks like that package was already on the image so it sounds like it needs
a different driver. Too bad.

khaltis, 2023-08-01:
@frutabruta did you ever resolve the touch issues with your screen? I appear to be having a similar issue. I can see that the touch device is detected but it is completely unresponsive.

SokolovJek, 2024-09-27:
cmd the 'sudo apt install xserver-xorg-input-libinput' help for me!