bigtreetech/CB1 #66: How to access GPIO via python

rollback91, 2022-11-23:
Hi, I've been trying to port a project I did for the raspberry pi to this new board. Unfortunately I got stuck when trying to controll the GPIO pins, as I was using the gpiozero library and it doesn't seem to support this board.

Do you know which python library I can use to controll the GPIO pins with the CB1?

MichaelMichaelMichaelMichaelMichael, 2023-11-23:
I have the same issue:
```
root@BTT-CB1:~# apt install python3-dev python3-rpi.gpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.9.2-3).
The following NEW packages will be installed:
  python3-rpi.gpio rpi.gpio-common
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 28.6 kB of archives.
After this operation, 98.3 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 rpi.gpio-common arm64 0.7.0-0.2+b2 [6,708 B]
Get:2 http://deb.debian.org/debian bullseye/main arm64 python3-rpi.gpio arm64 0.7.0-0.2+b2 [21.9 kB]
Fetched 28.6 kB in 0s (108 kB/s)            
Selecting previously unselected package rpi.gpio-common:arm64.
(Reading database ... 58153 files and directories currently installed.)
Preparing to unpack .../rpi.gpio-common_0.7.0-0.2+b2_arm64.deb ...
Unpacking rpi.gpio-common:arm64 (0.7.0-0.2+b2) ...
Selecting previously unselected package python3-rpi.gpio.
Preparing to unpack .../python3-rpi.gpio_0.7.0-0.2+b2_arm64.deb ...
Unpacking python3-rpi.gpio (0.7.0-0.2+b2) ...
Setting up rpi.gpio-common:arm64 (0.7.0-0.2+b2) ...
Setting up python3-rpi.gpio (0.7.0-0.2+b2) ...
root@BTT-CB1:~# /usr/bin/python3 /usr/local/bin/shutdown_button.py
Traceback (most recent call last):
  File "/usr/local/bin/shutdown_button.py", line 1, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
root@BTT-CB1:~# head -n 1 /usr/local/bin/shutdown_button.py 
import RPi.GPIO as GPIO

```
So, obviously, the message `This module can only be run on a Raspberry Pi!` tells me that this method is not working here. 
Which library will work in python3 in order to work with the GPIO pins?

Pneumanifest, 2023-11-24:
the CB1 is not a Raspberry pi clone, you can't use things designed for RPi. Try gpiod 
https://pypi.org/project/gpiod/

MichaelMichaelMichaelMichaelMichael, 2023-11-24:
gpiod works really fine! Good hint! Plus, it has some useful [examples](https://github.com/hhk7734/python3-gpiod/tree/main/py_src/gpiod/test). I need to convert my stuff though, but that's not too much to do. Thanks!

rollback91, 2023-11-24:
> the CB1 is not a Raspberry pi clone, you can't use things designed for RPi. Try gpiod https://pypi.org/project/gpiod/

Thanks I'll give it a try! At the moment I'm just using a custom board's gpio controlling it via serial uart