bigtreetech/CB1 #69: not connecting to network

astranberg, 2022-11-30:
Hello all,

I am using the latest version, v2.2.1. I flashed a fat32 formatted SD card with the file. Ejected it, plugged it back in, edited the system.cfg file with my wifi ssid and password, saved it, and then ejected it and placed it into the SD card labeled `CARD` on my manta MP8. I have VUSB shorted, and plugged the USB-C into the MP8. The red light turns onto the MP8 indicating it has power, and the green light on the CB1 turns on indicating it has power as well.

I have alternatively tried connecting a wifi USB dongle, and tried ethernet connection. I don't know if either of those require further configuration to work - but they didn't by just plugging them in.

Appreciate the help. This is very frustrating.

airbornetrooper82573, 2022-11-30:
I have mine configured and it has a very hard time getting wifi connection. I have the antenna sticking out of the skirt of my Voron and 9/10 times it won't connect. I ended up running a cat6 drop to my printer and have it hard wired in.

astranberg, 2022-11-30:
> I have mine configured and it has a very hard time getting wifi connection. I have the antenna sticking out of the skirt of my Voron and 9/10 times it won't connect. I ended up running a cat6 drop to my printer and have it hard wired in.

Thank you for the workaround. Does ethernet work by just plugging it in, or do you need to configure anything? My ethernet does not seem to work either.

airbornetrooper82573, 2022-11-30:
Mine works fine dhcp. There’s been an update to the Kernan and image so you can try that. It’s supposed to fix Wi-Fi issues but hasn’t done anything for me. It came out like 20 days ago or so

astranberg, 2022-11-30:
Yeah I'm already using v2.2.1

airbornetrooper82573, 2022-11-30:
That’s unfortunate. I’m on a replacement board and CB1 myself as both were dead on arrival. Kinda wish I went with the new Octopus instead but wanted to free up a pi4 for another project

Digitalpeer1, 2023-01-06:
I am having the same problem

astranberg, 2023-01-06:
The problem for me ended up being bad manta MP8.

Thanks,
Adam

Digitalpeer1, 2023-01-07:
oh, ok I emailed BTT to see if they will help

Digitalpeer1, 2023-04-18:
This was the issue Thanks