bigtreetech/CB1 #71: wlan0 not working

jpmini, 2022-12-03:
wlan0 not working

wlan0 doesn't work.

What was working suddenly stopped working. It's working on eth0, but when I check it, wlan0 is not recognized.

It doesn't work even if you drop the version.
I want you to tell me what to do...

[image]

dannyguana, 2023-04-02:
I have the same issue on 2.3.2 when i conect to usb-c in my manta m4p wifi works, and when i connect to 24 only detect ethernet wifi is not working

dkt0591, 2023-04-24:
same here, sigh, what a annoying about this

dkt0591, 2023-04-25:
update:

> same here, sigh, what a annoying about this

update: i changed power adapter, and it work again, that weird because previous release working fine with my old adapter.

Pneumanifest, 2023-04-25:
@dkt0591 what power supply was it that didn't let it work?

dkt0591, 2023-04-26:
> @dkt0591 what power supply was it that didn't let it work?

@Pneumanifest both are noname 3A phone charger

Pneumanifest, 2023-04-26:
> both are noname 3A phone charger
>

That's a problem. Phone chargers are not designed to be a power supply. They are charging circuits designed wo maintain battery health. They will reduce output when put under high load. Newer ones will even provide different voltage or shutdown depending on what it's attached to. You need to get an actual power supply like a raspberry pi power brick or a 5V (3A min.) PSU.