bigtreetech/CB1 #74: Manta M8P, CB1 and PanelDue via UART

tekknix, 2022-12-12:
Hi, 

i would like to connect my 7" PanelDue to one of CB1 UART's.

But i've no idea how to set UART1 TX and RX to be enabled on this Allwinner-Thing. 

With a CM4 it works, there i know how to do the Overlay and settings.

The standard UART(0) is used for the communication between the CB1 and Manta. Therefore i'm missing at least an additional UART(1).

Can someone bring some light into the darkness?

Regards
MoS

Townie-au, 2024-06-28:
Did you end up working this out?

frutabruta, 2024-07-08:
> The standard UART(0) is used for the communication between the CB1 and Manta

That is not true. CB1 communicates with manta over internal USB. The UART(0) is used for serial access to console.

You can use the serial port for something else, but you have to enable it in BoardEnv.txt, as mentioned on the [homepage](https://github.com/bigtreetech/CB1) of this repository:

>The default value is console=display, This means that the UART0 of CB1 is used as the debugging port by default. We can use MobaXterm to connect to CB1 by UART0 and debug. If klipper wants to use UART0 to control the motherboard, we need to set it to console=serial, now klippe can use UART0 as /dev/ttyS0.