bigtreetech/CB1 #75: What am i doing wrong

bombsquad85, 2022-12-12:
I downloaded the latest kernel but I get this error.

Also do you need a specific mouse/keyboard? I'm new to this. The mouse I'm using (wired optical) doesn't show a cursor and what I end up clicking on i
[image]
s random

bombsquad85, 2022-12-13:
I am also getting this message when I flash the latest kernel
[image]

frutabruta, 2022-12-13:
> I downloaded the latest kernel but I get this error.
> 
> Also do you need a specific mouse/keyboard? I'm new to this. The mouse I'm using (wired optical) doesn't show a cursor and what I end up clicking on i [image] s random

This error is normal, you have to create your own printer.cfg.

frutabruta, 2022-12-13:
> I am also getting this message when I flash the latest kernel [image]

This is normal too, the file is called system.cfg on CB1, so you cannot use raspberry pi imager to set WiFi. You need to edit the file manually on the SD card.

bombsquad85, 2022-12-14:
> > I downloaded the latest kernel but I get this error.
> > Also do you need a specific mouse/keyboard? I'm new to this. The mouse I'm using (wired optical) doesn't show a cursor and what I end up clicking on i [image] s random
> 
> This error is normal, you have to create your own printer.cfg.

is there a link to creating your own printer.cfg? i am very new to this. i finally got the image up and running, i had/have octoprint installed as well as klipper i guess but i cant get back into octoprint. so i guess i'll focus on klipper