bigtreetech/CB1 #78: Where is the i2c bus on the GIPIO ports?

alexMamaevs, 2022-12-15:
Hi,
Trying to connect ds1307 rtc module to CB1+PI4B
The overlays dtb file says that they are on the pins i2c0 "PI6", "PI7", i2c3 "PH4", "PH5".
i2c1 pins are missing.
Where are these pins on the board?

kostake2011, 2023-01-02:
Also very interesting. Failed to connect screen SSD1306 by i2c

pringsen, 2023-03-01:
Same here - i am using a CB1 for a non 3D printer related project and fail to find/use the i2c bus. Biqu please advise/help. :)

ifreislich, 2023-03-26:
According to the circuit diagram for the CB1 the I2C pins are not connected.  See sheet 9 and 10 of the schematic to see what's actually connected to the board interconnect.  I discovered this trying to make an MPU6050 work for the resonance tester in Klipper.

DrumClock, 2023-11-06:
....they are beautiful E-Waste, nothing more....

max-davletchin, 2024-09-24:
I agree with @DrumClock. They even have formal and incompetent support, which approves that their boards can only be used in 3D printers. Facepalm...