bigtreetech/CB1 #79: Any support for DPI displays?

billgeek, 2022-12-17:
I recently purchased the CB1 and Pi Adapter along with this DPI display from BTT:
https://biqu.equipment/collections/lcd/products/bigtreetech-pi-tft43-dip-screen

Now that support has been added for GPIO, is there any way to have the above DPI display working on the CB1? I did manage to get it working with my RPi 3A+ using `dtoverlay` config as per the [instructions](https://github.com/bigtreetech/TFT43-DIP), but I'm a Linux noob and have no idea how to perform the same config on the CB1. (there is no `/boot/config.txt` file?)

Untitled86, 2022-12-20:
I'm waiting on this feature.

emersonsc, 2022-12-26:
This will NOT be a feature available for the CB1 period. Its not a software issue, its a hardware "issue"

billgeek, 2022-12-26:
> This will NOT be a feature available for the CB1 period. Its not a software issue, its a hardware "issue"

What "hardware issue" are you referring to and is there any definitive statement from BTT / BIQU supporting this claim of hardware limitations? Not looking to "start something", just genuinely curious.

I'm no expert, but I've worked with quite a few hardware platforms and have never seen any piece of hardware that's fundamentally incompatible with another...

emersonsc, 2022-12-27:
The product pages for the PI4B, Manta 4, and Manta 8 CLEARLY state the CSI & DSI ports aren't compatible with the CB1. I would say that's a definitive statement from Biqu...

frutabruta, 2022-12-27:
> The product pages for the PI4B, Manta 4, and Manta 8 CLEARLY state the CSI & DSI ports aren't compatible with the CB1. I would say that's a definitive statement from Biqu...

Yes, DSI is not supported on Allwinner H616 and is not connected to anything on CB1. But OP asked about DPI, which is not the same.

DrumClock, 2023-08-23:
> The product pages for the PI4B, Manta 4, and Manta 8 CLEARLY state the CSI & DSI ports aren't compatible with the CB1. I would say that's a definitive statement from Biqu...

Hi,
please send a link to that statement.
Thank you

billgeek, 2023-08-23:
> > The product pages for the PI4B, Manta 4, and Manta 8 CLEARLY state the CSI & DSI ports aren't compatible with the CB1. I would say that's a definitive statement from Biqu...
> 
> Hi, please send a link to that statement. Thank you

I believe @emersonsc was referring to this:
https://biqu.equipment/en-fr/collections/cb1-family/products/pi4b-adapter-v1-0

[image]

DrumClock, 2023-08-23:
hmm... well, they're nice scumbags.
It's easy to overlook and you think you're buying a full-fledged replacement for the Raspberry CM4.

Pneumanifest, 2023-08-24:
> hmm... well, they're nice scumbags. It's easy to overlook and you think you're buying a full-fledged replacement for the Raspberry CM4.

There's also this on the same page in the features highlights. 
[image]

And nowhere does it state that it's a full-fledged replacement for a CM4. It's marketed, fairly clearly as an alternative option.  They even published a comparison chart showing it's not a copy. 
[image]

DrumClock, 2023-08-24:
sorry, but if the CB1 mod doesn't support DSI and CSI output, it's a big mistake.

In that case, I see no reason to use the PI4B adapter and MANTA boards. It's basically useless.

Pneumanifest, 2023-08-24:
> 

It's not a mistake that the cb1 doesn't support DSI CSI. It's likely a part of keeping the device in a certain price range. More kernel  development and added hardware would make it more expensive and an affordable alternative was the goal.
The Pi4b adapter and Manta boards are designed to also support the raspberry pi CM4. Compute module cross compatibility is far from useless. 
Regardless, if it's not meeting your needs then that is a valid reason to not purchase.

emersonsc, 2023-08-24:
> hmm... well, they're nice scumbags. It's easy to overlook and you think you're buying a full-fledged replacement for the Raspberry CM4.

How are they scum bags? Its not their fault you didnt bother to read. The documentation and website is VERY clear about the CB1 not supporting those features. 

> sorry, but if the CB1 mod doesn't support DSI and CSI output, it's a big mistake.
> 
> In that case, I see no reason to use the PI4B adapter and MANTA boards. It's basically useless.

How exactly is it a big mistake...? I run 4 printers with the Manta boards and a CB1. They work just fine without DSI and CSI. It sounds like you're more butthurt because YOU screwed up and didn't read the very clear data about what you bought and are now just being sour.

DrumClock, 2023-08-25:
> Manta boards and a CB1. They work just fine without DSI and CSI.

yes, you can easily have 100 printers with CB + MANTA and it will work fine. But if you need a display because neither LAN nor Wifi is available, you have a problem.

Yes, I didn't notice that DSI is not supported and since I have a TFT PI 50 with DSI, I was quite annoyed. Yes, the CM4 module would be fine (but it has been in short supply for a long time) and I was simply surprised that the DSI connector is brought out, but it is essentially unusable with CB1.

Now I will solve it with the help of SPI and TFT35.

Pneumanifest, 2023-08-25:
> > Manta boards and a CB1. They work just fine without DSI and CSI.
> 
> yes, you can easily have 100 printers with CB + MANTA and it will work fine. But if you need a display because neither LAN nor Wifi is available, you have a problem.
> 

There are HDMI touch screens and SPI that both work great with the CB1.