bigtreetech/CB1 #80: unable to edit files for wifi

jscancella, 2022-12-19:
I used raspberry pi imager to install `CB1_Debian11_bullseye_Klipper_kernel5.16_20221110.img` which was successful. However, after it was done installing there was no fat32 partition and windows (10) did not allow me to edit any files. I plan on using ethernet to get around this for now, but instructions should be updated because the current ones are not possible to follow.

chewy2420, 2023-01-14:
unplug and replug your drive. It ejects  it once its flashed, unfortuantly mine still wont connect after it is so *shrug* have a hard line around maybe.

Pneumanifest, 2023-01-14:
Not sure the process you are using to flash the image but the newest image works as described for me. Are you by chance clicking the format warning that pops up after the flash is finished? That should be ignored and not clicked. Also if you are using raspberry pi imager, do not use the option to setup wifi SSID and password and all that as it is not designed like the raspberry pi os.

chewy2420, 2023-01-14:
> 

Hi Pneumanifest, 
I didnt try to fix any error or such. I used etcher to flash it and believe im having the actual wifi issue from the first runs of boards, which I didnt think it effected the v2.1 boards. I was able to plug a screen and keyboard in and get to the network tab to see it says there is only eth0. Ive tried cb1 kernek v5.16 in standard and min flavors. I feel as tho I  might be just stuck with the cord, although not ideal but works. Thanksfor the reply and if there is anything else you can think of to try it would be greatfully appreciated -chewy

Pneumanifest, 2023-01-14:
Are you still not able to see and edit the system.cfg file with the computer? If you can't get to editing that with your wifi info it will not be able to connect to wifi.

chewy2420, 2023-01-14:
No I had no issue seting the network setting on the sd card, that was the op

Pneumanifest, 2023-01-14:
Oh ok. By chance did you use the stock notepad app to edit it? It has been known to add odd formating that can cause problems. It's best to use notepad++ or another code editor app.

chewy2420, 2023-01-14:
No I’ve been down that rabbit hole a few time so I’ve for the most part made notepad unusable on any pc that will touch a cfg in my home 😅 # never again.  Only notepad++ and vs code now :)

Doormat1, 2023-04-28:
The instructions in the manual dont work however this does - https://www.makeuseof.com/connect-to-wifi-with-nmcli/

chewy2420, 2023-04-28:
> The instructions in the manual dont work however this does - https://www.makeuseof.com/connect-to-wifi-with-nmcli/

So I’ve tried every way from Sunday but the actual WiFi card isn’t connected to the board it seems or it’s actually just completely dead, it works over Ethernet so I’ve haven’t tore it apart and put it under the microscope to really find out exactly what is the issue.

Joebus1, 2023-05-02:
While I can access my config file and input my SSID and Password, the device doesn't show up on my network. Ethernet works fine. Not sure what's wrong or how to trouble shoot it.

Doormat1, 2023-05-02:
Did you try following the instructions I linked and do it through terminal when connected over Ethernet

Joebus1, 2023-05-02:
> Did you try following the instructions I linked and do it through terminal when connected over Ethernet

I have em open in a new tab! I'll take a look tomorrow and check it out. Thank you!

fusiondew, 2023-10-15:
Had the same issue, just fixed it. If you open window's "Disk Management" program (What windows uses as a partitioning wizard) and find the drive, then right click the volume of the drive and select "Delete Volume" for all volumes, THEN use raspberry pi imager to install it onto the drive with no volumes, it should succeed. if the drive had any volume on it at all, even if it was a freshly formatted FAT32 volume, it would give me the same issue.

Also if you plug your drive in and windows doesn't detect it, if you go into the Disk Management program and assign a letter to the drive windows will then detect it.