bigtreetech/CB1 #81: USB Webcam not working

Reisbauer1489, 2023-01-01:
My Logitech C930e that I plugged into the printer was not showing anything at first, then I changed the device in crowsnest.conf and now its just showing "NO SIGNAL" on a black backround.
[image]
This is my crowsnest config:
[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log  # Default logfile in ~/klipper_logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_hurakan-if00 # See Log for available ...
resolution: 640x480                     # widthxheight format
max_fps: 60                            # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
Please help.

Apexpredation, 2023-01-01:
Device in Crowsnest is the camera not the MCU. 
Like this for example,
device: /dev/video0

Reisbauer1489, 2023-01-02:
It works! Thanks.

closed 2023-01-02 by Reisbauer1489

calmarc, 2024-03-01:
biqu@BTT-CB1:~$ v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.

These are the modules installed:

biqu@BTT-CB1:~$ lsmod
Module                  Size  Used by
zram                   32768  2
nls_iso8859_1          16384  1
8189fs               1187840  0
cfg80211              352256  1 8189fs
dw_hdmi_cec            16384  0
cdc_acm                36864  2
sun50iw9_codec         28672  3
panfrost               65536  1
gpu_sched              32768  1 panfrost
display_connector      20480  0
fuse                  126976  1
i2c_mv64xxx            24576  0

Is there anything missing?