bigtreetech/CB1 #82: UNABLE TO FIND MCU - PI4B + CB1

enricoxg, 2023-01-06:
Hello, 

I am having trouble finding the MCU ID through he USB port of the PI4b adapter. I have a USB stick that flashes blue when i plug it into a computer. However, when i plug it into the Pi4B the 4 usb ports, non of them turn on my usb stick! It seems like all 4 usb ports are broken, or not activated.

frutabruta, 2023-01-06:
Do you have access to SSH or a monitor with keyboard setup?
If so, type `lsusb` into terminal before and after pluging something into USB.

enricoxg, 2023-01-06:
> Do you have access to SSH or a monitor with keyboard setup? If so, type `lsusb` into terminal before and after pluging something into USB.

None of my usb devices show up in any of the 4 usb ports.

LucLN, 2023-01-14:
see #83 .. have you updated the OS to the latest version ?