bigtreetech/CB1 #84: TS35 / Generic SPI screen support?

tri4ng1e, 2023-01-19:
Hello!

Is there a possibility to connect [TS35 touch screen](https://github.com/makerbase-mks/MKS-TFT-Hardware/tree/master/MKS%20TS35-R/MKS%20TS35-R%20V2.0_002)? If I understand it correctly, it's a SPI-based screen, so in theory it should be compatible with CB1 since it's not a DPI one. I tried to follow a [guide made for raspberry pi](https://github.com/fbeltrao-br/mksts35) to set up KlipperScreen, but encountered some issues, mainly being:
1. In the guide they used a dtoverlays called `tinylcd35` and `touch`, but there are no such overlays in CB1 (it seems). I am not quite familiar with device tree overlays, but in my understanding they define a hardware device by its pins and features. Is it possible to make those overlays or copy them from rpi install? I think they will require some changes...
2. While building `fbcp`, I got some compile errors related with some headers that should be a part of couple packages, but they are not in the repos, it seems. I didn't pay much attention to this, since without P.1 it's useless anyway.

I've found 3 possibly interesting CB1 overlays called `tft35_spi`, `mcp2515` and `spi-spidev`, they all are related to SPI operation, but I'm afraid they will not work as expected in the guide in its current form. First one is made for [BTT's SPI screen](https://biqu.equipment/products/bigtreetech-tft35-spi-v2-1-touchscreen-io2can-module), which uses I2C touch control, and it seems TS35 uses SPI for touch as well, so I'm not sure.

Maybe somebody had some experience with all of those, or even connected a SPI screen to CB1?

Thank you.

emersonsc, 2023-01-24:
I have 3 of the new SPI screens with the IO2CAN board so I can make all my Manta Boards have CanBus and I notice ALL the new product they dropped right before the Chinese New Year got fresh new Github pages... except for those two devices. My products will be here in a few days (per tracking) and there is ZERO for support on how to use these.

As far as the mcp2515 goes, i believe thats for using the [RPI Can Hat](https://www.amazon.com/RS485-CAN-HAT-Long-Distance-Communication/dp/B07VMB1ZKH/ref=asc_df_B07VMB1ZKH/?tag=hyprod-20&linkCode=df0&hvadid=524643816709&hvpos=&hvnetw=g&hvrand=10126933267063842144&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1019336&hvtargid=pla-1225123461783&psc=1) with some slight modification (you have to jump a pin or two).

Eclypse3D, 2023-01-31:
The CB1 according to their site does support the screen but only v2.2 of the board, the older version must use the canbus MCP2515 I believe to use the screen...  If you look at the cfg files in the root of the sdcard firmware for the cb1 there is a config file where you uncomment out features to make use of either SPI, Canbus, both etc.

frutabruta, 2023-02-01:
> I have 3 of the new SPI screens with the IO2CAN board so I can make all my Manta Boards have CanBus and I notice ALL the new product they dropped right before the Chinese New Year got fresh new [...]

Those missing github pages have been just released:
https://github.com/bigtreetech/IO2CAN
https://github.com/bigtreetech/TFT35-SPI

emersonsc, 2023-02-01:
> Those missing github pages have been just released: https://github.com/bigtreetech/IO2CAN https://github.com/bigtreetech/TFT35-SPI

Yeah and if you noticed I already found them and posted an issue