bigtreetech/CB1 #89: CB1 reporting incorrect resolution to Klipperscreen

eboston, 2023-02-08:
Trying to get my klipper setup moved over to the Manta M5P with a CB1.  I have gotten most of it working, but can't get the proper resolution provided to Klipperscreen.  I am using a 7" 1024x600 HDMI Touch screen display.  When I start the system, I could see the splash screen was stretched vertically.   I was able to get the resolution set to the proper 1024x600 and the start screen shows properly now.  However, when I run Klipperscreen, the display is elongated. Checking the logs, it shows the system reported the resolution as 1024x768.

I added 

    extraargs=video=HDMI-A-1:1024x600-24@60

to the BoardEnv.txt file which fixed the resolution to the boot screen.  Based on the changes I did to get the display to work with the RPi4, I also added:

  hdmi_group=2
  hdmi_mode=87
  hdmi_cvt=1024 600 60 3 0 0 0

To the file.  Klipperscreen is still getting the wrong resolution.

Is there somewhere that I need to set the resolution at?

0V3RK7LL, 2023-03-07:
I just recently figured out the solution to this problem.  Under BoardEnv.txt, I added extraargs=video=HDMI-A-1:1024x600-24@60 like you did.
I had to SSH into the board and follow the klipperscreen instructions.

sudo nano /usr/share/X11/xorg.conf.d/90-monitor.conf

I pasted these instructions:

Section "Monitor"
    Identifier "DPI-1"
    # This identifier would be the same as the name of the connector printed by xrandr.
    # it can be "HDMI-0" "DisplayPort-0", "DSI-0", "DVI-0", "DPI-0" etc

    Option "Rotate" "left"
    # Valid rotation options are normal,inverted,left,right

    Option "PreferredMode" "1920x1080"
    # May be necesary if you are not getting your prefered resolution.
EndSection

I then changed a few things:

Section "Monitor"
    Identifier "HDMI-1"
    # This identifier would be the same as the name of the connector printed by xrandr.
    # it can be "HDMI-0" "DisplayPort-0", "DSI-0", "DVI-0", "DPI-0" etc

    #Option "Rotate" "left"
    # Valid rotation options are normal,inverted,left,right

    Option "PreferredMode" "1024x600"
    # May be necesary if you are not getting your prefered resolution.
EndSection

Saved and then exited. 
I restarted klipperscreen and the touch panel resolution was correct from then on in.

sudo service KlipperScreen restart

eboston, 2023-03-07:
Thanks for the info.  I ended up getting a BIQU HDMI7 display and it works without any changes so it looks like the other screens I have are not reporting the correct information.  I have a couple other machines I am going to update the display on and will give this info a try then.

bboybattlemage, 2023-03-29:
> I just recently figured out the solution to this problem. Under BoardEnv.txt, I added extraargs=video=HDMI-A-1:1024x600-24@60 like you did. I had to SSH into the board and follow the klipperscreen [...]

If this still working? Thought I had found a solution to my annoying problem, but completing your actions still had my Klipperscreen attempting to output at a funny resolution.

I have had some additional success forcing xrandr into a new display mode when klipperscreen starts. This gets me a clean and crisp image on the display, but the image is stretched in the vertical plane. Mucking around with the xrandr scale, I can get the whole klipperscreen dashboard in the display with working touch functions, however it does not fill the entire display, with a portion of the screen being blank.

I've spent the last few nights attempting to resolve this issue but still don't have a great solution... Thinking of ditching the CB1 and going with the older, more accessible raspberry Pi board...