bigtreetech/CB1 #92: create hotspot

jscancella, 2023-02-10:
Is there an officially supported way to create a hotspot on the CB1 image?  I tried following https://www.raspberryconnect.com/projects/65-raspberrypi-hotspot-accesspoints/157-raspberry-pi-auto-wifi-hotspot-switch-internet but it just fails to stay connected and goes into an endless loop trying to connect.

jscancella, 2023-02-11:
I tried to follow https://computingforgeeks.com/create-wi-fi-hotspot-on-linux/ which should work since you are using networkmanager for the wifi, but as you can see below I am getting an error when it tries to bring the wifi hotspot up. This is using the https://github.com/bigtreetech/CB1/releases/tag/V2.3.0 minimal image.

```
biqu@BTT-CB1:~$ ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether e6:d3:36:f4:bf:7e brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DORMANT group default qlen 1000
    link/ether 94:a4:08:b3:0e:e2 brd ff:ff:ff:ff:ff:ff
biqu@BTT-CB1:~$ IFNAME="wlan0"
biqu@BTT-CB1:~$ CON_NAME="RatOS"
wifi ifname $IFNAMbiqu@BTT-CB1:~$ nmcli con add type wifi ifname $IFNAME con-name $CON_NAME autoconnect yes ssid $CON_NAME
nmcli con modify $CON_NAME wifi-sec.key-mgmt wpa-psk
nmcli con modify $CON_NAME wifi-sec.psk "raspberry"

Connection 'RatOS' (f6f294c4-41a0-458c-8b37-f3d1f59c20e1) successfully added.
biqu@BTT-CB1:~$
biqu@BTT-CB1:~$ nmcli con modify $CON_NAME wifi-sec.key-mgmt wpa-psk
biqu@BTT-CB1:~$ nmcli con modify $CON_NAME wifi-sec.psk "raspberry"
biqu@BTT-CB1:~$
biqu@BTT-CB1:~$ nmcli con up $CON_NAME
Error: Connection activation failed: The Wi-Fi network could not be found
Hint: use 'journalctl -xe NM_CONNECTION=10cc53e5-3b4d-44d8-bd27-28a9292ac406 + NM_DEVICE=wlan0' to get more details.
biqu@BTT-CB1:~$ sudo journalctl -xe NM_CONNECTION=10cc53e5-3b4d-44d8-bd27-28a9292ac406 + NM_DEVICE=wlan0
Feb 07 10:26:49 BTT-CB1 NetworkManager[1019]: <info>  [1675765609.6874] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/3)
Feb 07 10:26:49 BTT-CB1 NetworkManager[1019]: <info>  [1675765609.6941] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Feb 07 10:26:49 BTT-CB1 NetworkManager[1019]: <info>  [1675765609.9009] device (wlan0): supplicant interface state: internal-starting -> disconnected
Feb 07 10:26:49 BTT-CB1 NetworkManager[1019]: <warn>  [1675765609.9159] sup-iface[6032f6448f89e0f7,0,wlan0]: call-p2p-cancel: failed with P2P cancel failed
Feb 07 10:26:49 BTT-CB1 NetworkManager[1019]: <info>  [1675765609.9163] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.1845] device (wlan0): Activation: starting connection 'ZYIPTest' (e7391fed-5d7e-49c9-b3d7-42b347bef8f0)
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.1892] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.1985] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2020] device (wlan0): Activation: (wifi) access point 'ZYIPTest' has security, but secrets are required.
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2034] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2223] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2260] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2300] device (wlan0): Activation: (wifi) connection 'ZYIPTest' has security, and secrets exist.  No new secrets needed.
Feb 07 10:26:54 BTT-CB1 NetworkManager[1019]: <info>  [1675765614.2727] device (wlan0): supplicant interface state: disconnected -> scanning
Feb 07 10:27:19 BTT-CB1 NetworkManager[1019]: <warn>  [1675765639.4804] device (wlan0): Activation: (wifi) association took too long, failing activation
Feb 07 10:27:19 BTT-CB1 NetworkManager[1019]: <info>  [1675765639.4806] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Feb 07 10:27:19 BTT-CB1 NetworkManager[1019]: <warn>  [1675765639.4888] device (wlan0): Activation: failed for connection 'ZYIPTest'
Feb 07 10:27:19 BTT-CB1 NetworkManager[1019]: <info>  [1675765639.4895] device (wlan0): supplicant interface state: scanning -> disconnected
Feb 07 10:27:19 BTT-CB1 NetworkManager[1019]: <info>  [1675765639.4941] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Feb 11 15:16:51 BTT-CB1 NetworkManager[1019]: <info>  [1676128611.2837] device (wlan0): Activation: starting connection 'RatOS' (10cc53e5-3b4d-44d8-bd27-28a9292ac406)
Feb 11 15:16:51 BTT-CB1 NetworkManager[1019]: <info>  [1676128611.2862] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Feb 11 15:16:51 BTT-CB1 NetworkManager[1019]: <info>  [1676128611.2938] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb 11 15:16:51 BTT-CB1 NetworkManager[1019]: <info>  [1676128611.2961] device (wlan0): Activation: (wifi) access point 'RatOS' has security, but secrets are required.
[... cut ...]
Feb 11 15:24:01 BTT-CB1 NetworkManager[1019]: <info>  [1676129041.3254] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Feb 11 15:24:01 BTT-CB1 NetworkManager[1019]: <warn>  [1676129041.3339] device (wlan0): Activation: failed for connection 'RatOS'
Feb 11 15:24:01 BTT-CB1 NetworkManager[1019]: <info>  [1676129041.3345] device (wlan0): supplicant interface state: scanning -> disconnected
Feb 11 15:24:01 BTT-CB1 NetworkManager[1019]: <info>  [1676129041.3381] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
biqu@BTT-CB1:~$
```

jscancella, 2023-02-11:
I also tried this, but again I get the behavior that it connects but then immediately disconnects
```
echo "interface=wlan0
driver=nl80211
ssid=RatOS
hw_mode=g
channel=6
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=raspberry
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP" | sudo tee /etc/hostapd/hostapd.conf

nmcli connection add \
    con-name RatOS \
    ifname wlan0 \
    type wifi \
    autoconnect yes \
    ipv4.method shared \
    802-11-wireless.band bg \
    802-11-wireless.mode ap \
    802-11-wireless.ssid "RatOS" \
    802-11-wireless-security.key-mgmt wpa-psk \
    802-11-wireless-security.psk "raspberry"
nmcli con up RatOS
```

HelgeKeck, 2023-02-12:
yeah, i have the exact same iusse

mmuellerphoto, 2023-02-12:
Same here! Would be very helpful to have a hotspot.

Flavourdynamics, 2023-02-13:
This would be great.

jscancella, 2023-02-13:
For people looking for an answer in the future, https://github.com/oblique/create_ap seems to work

kieraneglin, 2023-02-18:
@jscancella what command did you use to get this running and what is your use case? I want my CB1 to act as a wifi AP without internet sharing but I keep getting "Your adapter can not be a station" errors

jscancella, 2023-02-19:
@kieraneglin using it for Rat-OS. It is used to setup the system on first boot when it isn't connected to the local wifi yet. I was talking with a BTT tech and they said to use create_ap which worked

kieraneglin, 2023-02-19:
@jscancella thank you! That helps a lot. Is there any chance you remember what command you are issuing on startup?

jscancella, 2023-02-19:
I changed the config file
https://github.com/Rat-OS/RatOS/blob/v2.x/src/modules/hotspot_cb1/filesystem/root/etc/create_ap.conf
And then ran `sudo systemctl enable create_ap`
See the whole install script here: https://github.com/Rat-OS/RatOS/blob/v2.x/src/modules/hotspot_cb1/start_chroot_script

closed 2023-02-19 by jscancella